hultenvp/solis-sensor #289: Migration of Solis and Sofar inverters

kazikskrz, 2023-06-01:
Hello. 
I own 2 plants/inverters: Solis, Sofar. Been using platform2 integration for both till 30.05.2023. Then executed migration process and as expected only one plant running Solis plant has got migrated to SolisCloud. I asked for API and applied it on the portal and I can integrate it to HA. Sofar remained in Solarman platform2 but integration stopped working. Can you tell why?
Both plants still exist in paralel in SolarMan.
In HA my Solis "sensor.solis_energy_today" had a different name "sensor.inverter_energy_today". I have renamed it in SolisCloud for better naming convention. How can I migrate to HA to keep statistics of old sensor name? 
Please advise. Thank you.

simonjgreen, 2023-06-02:
Just to jump on, I have a very similar challenge although my two inverters are both Solis. They were on the old integration until the 30th cutover. I can now see both dataloggers but only 1 inverter in the portal (though that looks to be a Solis problem and I've opened a ticket). I've got the inverter added to the integration in HA however I don't want to throw away my historic data logged out of the old API.

kazikskrz, 2023-06-05:
Looks exactly the same as ginlong, to be honest. For HA I started using https://github.com/StephanJoubert/home_assistant_solarman

This is great news. Can you tell how to migrate from Platform2 into this Solarman and keep Energy statistics with the same entity names?

closed 2023-06-10 by kazikskrz