hultenvp/solis-sensor #290: No data since m.ginlong closed, no API from soliscloud

maciejka104, 2023-06-01:
Since m.ginlong closed I can't configure soliscloud access. 
I can't get API from solis. 
Any idea how to get solscloud api? 
I got error message"No authority"

**Versions**
HA Version an deployment - 2023.5.4
HACS version 1.32.1
Integration version 3.3.3

**Debug logs**
Add to configuration.yaml:
```

    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

jaypikay, 2023-06-02:
I am not sure, but you might have to do step 1 of this https://github.com/hultenvp/solis-sensor/#soliscloud. "Submit a [service ticket](https://solis-service.solisinverters.com/support/solutions/articles/44002212561-api-access-soliscloud) and wait till it is resolved.".

maciejka104, 2023-06-02:
Solved

closed 2023-06-02 by maciejka104