hultenvp/solis-sensor #293: Multiple inverters grouped into one

henrikeri, 2023-06-12:
**Describe the bug**
For sites that has more than one inverter, all inverters are currently grouped into one. This makes it hard to keep an overview of what is what, and as such requires a lot of manual correlation between the web page and HA. 

**Config**
Using the new soliscloud with API key

**Versions**
All updated to latest NON beta versions

**Screenshots**
Multiple inverters grouped into 1:

[image]

henrikeri, 2023-06-13:
No errors from that I can tell in the log, in fact when resetting after enabling debugging it shows the following:

2023-06-13 11:10:41.708 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['aaa', 'bbb', 'ccc']
2023-06-13 11:10:41.708 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 'ddd'

hultenvp, 2023-06-19:
Do you see the separate inverters as devices in the device overview?
(Settings->Services and services->Solis Inverter). 
The idea is it shows each individual inverter SN as a device.

henrikeri, 2023-06-19:
Hi,

Funny enough, I see two of three. But only one of the devices ( serial numbers ) has data in it

swests, 2023-06-20:
Similar. I see both inverters after a reload but all the entities are under one inverter. The other has no entities.

hultenvp, 2023-06-20:
That's odd.

The device registry is built from the device_info method on the sensor class. Inverter serial and platform are the unique identifiers I use in the device identifier.
It should group just fine, it does group nicely for me (but I have one inverter, but for testing purposes used it on both platforms). So either the sensor gets the wrong inverter serial or the mechanism I use to create a unique identifier for the device doesn't work as I understand it works. Give that it works for me I go for the first option for now.
Can you run with debug on and reload the integration? I'm looking for a list of debug calls saying "Creating xxxx (yyyy)" where yyyy is the inverter serial. 
You should see all sensors created for all 3 registered inverters.
Please drop the debug log here.
[image]

swests, 2023-06-20:
I have a debug log in #285 . As these seem similar issues is it worth merging the two issues?

henrikeri, 2023-06-23:
`023-06-23 18:01:54.186 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":{"accumulatedConsumption":6.78,"accumulatedConsumptionLastHour":0,"accumulatedCost":9.133106,"accumulatedProduction":54.997528,"accumulatedProductionLastHour":0.037528,"accumulatedReward":59.416945,"averagePower":376,"currency":"NOK","currentL1":null,"currentL2":null,"currentL3":null,"lastMeterConsumption":32685.83,"lastMeterProduction":8498.65,"maxPower":2853,"minPower":0,"power":0,"powerFactor":null,"powerProduction":null,"powerReactive":null,"signalStrength":null,"timestamp":"2023-06-23T18:01:52.500+02:00","voltagePhase1":null,"voltagePhase2":null,"voltagePhase3":null}}}}
2023-06-23 18:01:55.216 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': False, 'use_energy_today_from_plant': False}
2023-06-23 18:01:55.220 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solis
2023-06-23 18:01:55.223 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2023-06-23 18:01:55.223 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2023-06-23 18:01:56.546 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2023-06-23 18:01:56.546 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['100504021B290022', '010CF1216300005', '010CF1216300029']
2023-06-23 18:01:56.546 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 100504021B290022
2023-06-23 18:01:56.638 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":{"accumulatedConsumption":6.78,"accumulatedConsumptionLastHour":0,"accumulatedCost":9.133106,"accumulatedProduction":54.997528,"accumulatedProductionLastHour":0.037528,"accumulatedReward":59.416945,"averagePower":376,"currency":"NOK","currentL1":null,"currentL2":null,"currentL3":null,"lastMeterConsumption":32685.83,"lastMeterProduction":8498.65,"maxPower":2853,"minPower":0,"power":0,"powerFactor":null,"powerProduction":null,"powerReactive":null,"signalStrength":null,"timestamp":"2023-06-23T18:01:55.000+02:00","voltagePhase1":null,"voltagePhase2":null,"voltagePhase3":null}}}}
2023-06-23 18:01:58.737 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2023-06-23 18:01:58.737 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 100504021B290022
2023-06-23 18:01:59.411 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":{"accumulatedConsumption":6.78,"accumulatedConsumptionLastHour":0,"accumulatedCost":9.133106,"accumulatedProduction":54.997528,"accumulatedProductionLastHour":0.037528,"accumulatedReward":59.416945,"averagePower":376,"currency":"NOK","currentL1":null,"currentL2":null,"currentL3":null,"lastMeterConsumption":32685.83,"lastMeterProduction":8498.65,"maxPower":2853,"minPower":0,"power":0,"powerFactor":null,"powerProduction":null,"powerReactive":null,"signalStrength":null,"timestamp":"2023-06-23T18:01:57.500+02:00","voltagePhase1":null,"voltagePhase2":null,"voltagePhase3":null}}}}
2023-06-23 18:02:00.030 INFO (MainThread) [gql.transport.websockets] >>> {"type": "ping"}
2023-06-23 18:02:00.078 INFO (MainThread) [gql.transport.websockets] <<< {"type":"pong"}
2023-06-23 18:02:00.265 INFO (SyncWorker_7) [custom_components.icloud3.device_tracker] Karoline (iPhone) Results: From-Home > CurrZone-Home, GPS-(58.926636, 5.614242/8.514426531661275m), Interval-2 hrs (3-InHomeZone), NextUpdt-20:02:00, LocationTime-17:50:07 (11.9 min ago), OldLocThreshold-3 min, Battery-8%
2023-06-23 18:02:00.422 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2023-06-23 18:02:00.422 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['100504021B290022', '010CF1216300005', '010CF1216300029']
2023-06-23 18:02:00.422 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 100504021B290022
2023-06-23 18:02:00.907 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2023-06-23 18:02:00.908 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 010CF1216300005
2023-06-23 18:02:02.525 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":{"accumulatedConsumption":6.78,"accumulatedConsumptionLastHour":0,"accumulatedCost":9.133106,"accumulatedProduction":55.000911,"accumulatedProductionLastHour":0.040911,"accumulatedReward":59.420452,"averagePower":376,"currency":"NOK","currentL1":-7.3,"currentL2":-2.5,"currentL3":4.6,"lastMeterConsumption":32685.83,"lastMeterProduction":8498.65,"maxPower":2853,"minPower":0,"power":0,"powerFactor":null,"powerProduction":1218,"powerReactive":0,"signalStrength":null,"timestamp":"2023-06-23T18:02:00.000+02:00","voltagePhase1":239.7,"voltagePhase2":234.7,"voltagePhase3":234.9}}}}
2023-06-23 18:02:03.987 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":{"accumulatedConsumption":6.78,"accumulatedConsumptionLastHour":0,"accumulatedCost":9.133106,"accumulatedProduction":55.000911,"accumulatedProductionLastHour":0.040911,"accumulatedReward":59.420452,"averagePower":375.9,"currency":"NOK","currentL1":null, [... cut ...]
2023-06-23 18:02:25.098 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute socDischargeSet for inverter 100504021B290022
2023-06-23 18:02:25.098 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:01:00, at 2023-06-23 16:03:25.098839+00:00
2023-06-23 18:02:25.823 WARNING (MainThread) [zigpy.zcl] [0x01E1:1:0xfeed] Unknown cluster command 42 b''
2023-06-23 18:02:27.134 INFO (MainThread) [gql.transport.websockets] <<< {"id":"1","type":"next","payload":{"data":{"liveMeasurement":`

hultenvp, 2023-07-10:
I can (finally) reproduce the issue, no solution yet

swests, 2023-07-12:
Thanks for the fix - my setup is now showing correctly!