hultenvp/solis-sensor #294: Cannot Login with provided URL and Credentials

Jonnehs, 2023-06-14:
**Describe the bug**
Unable to login, but the test script does connect to the API and work

**Versions**
HASS Version 2023.06.1
HACS version 1.32.1
Integration version 3.3.3

**Debug logs**
couldnt see any generated, except the error that the integration has no plants configured (duh)

**Screenshots**
[image]
[image]

advertisingfree, 2023-06-16:
This integration is not working at all. I have been wasting so much time.

Luke-Beck, 2023-06-16:
Just commenting to say I'm getting the same issue.
I've recently migrated to SolisCloud from the old platform, if that's a common denominator.

FGOD1983, 2023-06-16:
also migrated from the old platform to soliscloud and having the same issue. Then again, I'm also having issues getting the verification mail from Soliscloud which only worked once and thank god I made a picture of the key and secret so I at least am able to try login later on...

But yeah, it seems that something changed on the api and made it work in a different way. 

did somebody split through the code and figured out what the full api url is that the code is doing to get info? because whatever I try results in Request method not supported:

```
<ResponseResult>
<Success>false</Success>
<Code>1</Code>
<Msg> Request method not supported http://www.soliscloud.com:13333/v1/api/inverterDetail</Msg>
<Data/>
<CsrfToken/>
</ResponseResult>
```

FGOD1983, 2023-06-16:
I finally found the issue!

Upon the migration the Plant ID is left empty. You need to edit the plant in order to let it generate an ID, and after that I was able to login. 

Thx HamsenBender for the solution: https://github.com/hultenvp/solis-sensor/issues/232#issuecomment-1566006420 :)

steoj, 2023-06-16:
Edit the plant in order to let it generate an ID fixed the authentication issue for me too.

Jonnehs, 2023-06-19:
Mine did have an ID, and the test worked, but editing the plant name also fixed this for me as well

Worth adding to the wiki @hultenvp ?

closed 2023-06-19 by Jonnehs