hultenvp/solis-sensor #296: Weird impossibly high spike in "Solar production" on Home Assistant energy dashboard

ncot-tech, 2023-06-16:
I don't know if this is an issue with this integration or with Home Assistant, but recently I've noticed my solar production graph has a weird and impossibly high spike in the morning...

[image]

(I mean, I'd like to somehow magically make that much energy in the morning from my 3kW setup, I could charge my car with it! but I think something's not quite right)

However, the sensor it's using looks fine...

[image]

And it matches what Solis Cloud thinks

[image]

So the data seems to be coming out of Solis Cloud properly, but for some reason HA is going a bit strange. Any ideas how I can debug this? My energy dashboard is set up like this:

[image]

bluemoon83, 2023-06-16:
I've had similar issues over the couple of weeks but mine does it between 2300-0000 and the spike is the size of solar production for that day. 
[image]
[image]

Not sure why it is happening this post https://community.home-assistant.io/t/correcting-incorrect-energy-statistics/429720 does show how you can remove them, with relative ease, afterwards so it doesn't distort your stats.

jahpster, 2023-06-19:
I have the same problem since the 6 of June.

hultenvp, 2023-06-20:
See also #273 

@ncot-tech can you also check the history of the "Solis daily grid energy purchased"  and "Solis daily on grid energy". Wouldn't be surprised if anything is wrong with those. 
Soliscloud API is extremely unreliable. This is likely one of those examples, I don't think this is an energy dash issue. If I know which fields misbehave I might be able to apply some filtering to mitigate the issues, but no guarantees.

wesjuhdabomb, 2023-06-26:
@hultenvp. Same issue here, however I only see 0 amounts on the "Solis daily grid energy purchased" and "Solis daily on grid energy".

What I do see on the days when it is spiking (not happening every day), "Solid Energy Today" jumps to 0 and back to the total production in the last hour before midnight. Hopefully this helps. In my case a filter for any production between 23:00 and 0:00 would eleminate the issue. 
[image]

bbartje, 2023-06-27:
> @hultenvp. Same issue here, however I only see 0 amounts on the "Solis daily grid energy purchased" and "Solis daily on grid energy".
>
> What I do see on the days when it is spiking (not happening [...]

That sounds like a perfect workaround, I have the same problem and don't want to keep cleaning up my stats.
How do you create such a filter?

chmutoff, 2023-06-27:
I have exactly the same issue. The counter in solis resets a 0:00 and this seems to break the stats in energy dashboard

hultenvp, 2023-06-28:
Others have reported these fluctuations on different moments during the day.
Filtering out could work under the following conditions:
* The integration has no ability to persist state, so you'll never be able to filter out if HA restarts just at the moment SC reports 0 kWh.
* There is already a filter to ignore any non-zero values when the inverter is still off. This is to filter out situations where energy is reset too late after midnight. This filter and the reset at midnight should continue to work.

The existing filter is in [service.py](https://github.com/hultenvp/solis-sensor/blob/ee36537f37e80d5593f73483ec45bcac230d48c1/custom_components/solis/service.py#L158)
A new one must be added that keeps track of previous values and ignores drops (any drop, even if it's not to 0 is not allowed if it's not the daily reset)

As said in different threads I have little time at the moment to work on the integration. I do welcome and will process community contributions.

wesjuhdabomb, 2023-07-04:
Not fully understanding the code, but I made a quick and dirty change in my homeassistant setup and it seems the spikes are gone ever since. The change was made 1wk ago and I used to see them every 2-3 days.

From my basic understanding the code, my reasoning was to ignore any values from polluting the data when the inverter is off, since there should not be any additional energy production with the inverter in OFF state. These lines were added after line 163 in service.py.

```
is_pm = datetime.now().hour > 12
if getattr(data, INVERTER_STATE) == 2 and is_pm:
   return
```

hultenvp, 2023-07-09:
Thanks, let's go for this fix for now. Not sure if it only appears during cases where the inverter is off, but we can refine later if necessary.

closed 2023-07-09 by hultenvp