hultenvp/solis-sensor #299: Can no longer get data or create new integration with Solis login

usbrit, 2023-07-05:
**Describe the bug**
About a month ago I lost all sensor data from the integration. I have tried re entering my login both via ginlong and soliscloud, neither work. 
I deactivated my API, reactivated and tried soliscloud access again, still will not let me setup the integration 

**Versions**
HA Version an deployment-2023.4.6
HACS version-1.32.1
Integration version-3.3.3

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

Thanks,
David

hultenvp, 2023-07-09:
Ginlong is eond of service and won't work anymore. 
For Soliscloud: please provide debug log, so I can have a look.

usbrit, 2023-07-09:
Fixed!
Signed in again today, all good. Sorry to have troubled you

MichalKurzewski, 2023-07-11:
so my issue was i blindly copied https://www.soliscloud.com:13333/ from api key retrieval where it should be https://www.soliscloud.com:13333 <- without the slash.

hultenvp, 2023-07-11:
Thanks for the update, closing

closed 2023-07-11 by hultenvp