hultenvp/solis-sensor #307: Time Measurements Received Date/Timestamp incorrect

viking2010, 2023-07-15:
Hi

After upgrading my HA to the latest version (Home Assistant 2023.7.2 / Supervisor 2023.07.1 / Operating System 10.3 / Frontend 20230705.1 - latest), and then upgrading to the latest version of the integration (v3.4.0), the value for sensor.solis_inverter_timestamp_measurements_received isn't showing correctly and the decimal place keeps moving.

When looking back at the history of the time stamp measurement you can see (on the image below) that just after 18:00 on 14th July is where I updated HA.

[image]

....After that you can see the spikes in the values. Here's a closer look at those from today:

[image]

What seems to happen is that the peaks are actually the correct values that should be there, and the troughs are when the values are incorrect. 

Looking at it from the Developer Tools, you can see further what's happening. 

```
Current UNIX Timestamp: {{ states('sensor.solis_inverter_timestamp_measurements_received')}}
Value as DD/MM/YYYY: {{ states('sensor.solis_inverter_timestamp_measurements_received') | float(0) | timestamp_custom ('%d/%m/%Y %H:%M:%S') }}
```

The first line gives the timestamp of 1689424.482395
This equates to: 20/01/1970 14:17:04

The decimal point is 4 places out and it should be 1689424482.39
The would be: 15/07/2023 13:34:42

I know in the latest version you increased the number of decimal places on some values, so could something be altering the timestamp field?

If I can provide any further details please let me know what you need.

Many thanks

hultenvp, 2023-07-16:
Hey @viking2010,

Thanks for spotting. I took in a PR for ingesting some grid sensor data from station detail. I now see this means post_process is called twice, resulting in the timestamp being corrected twice. I believe that's the cause.

viking2010, 2023-07-16:
Ahh ok cool. At least you know what the issue is which makes fixing it simpler! I threw in a template sensor to correct it but will look out for an update. If you want me to test any code before you release it, let me know.

hultenvp, 2023-07-17:
Oh, that would be great, because strange enough I cannot reproduce the issue.
I got a potential fix ready in the branch under this issue. If you could replace soliscloud_api.py with the version from this branch: https://github.com/hultenvp/solis-sensor/blob/307-time-measurements-received-datetimestamp-incorrect/custom_components/solis/soliscloud_api.py?

Hope this version works

viking2010, 2023-07-17:
Hi

I can confirm that the changes made in the new soliscloud_api.py have fixed the issue.  If this changes I'll let you know, but it's been running fine for the couple of hours.

Thanks for looking in to this and providing a fix.

hultenvp, 2023-07-30:
Thanks!
Closing

closed 2023-07-30 by hultenvp

viking2010, 2023-08-08:
> Thanks! Closing

@hultenvp Can you confirm if the above changes you made were incorporated in the latest versions?

hultenvp, 2023-08-24:
Was on holidays, missed this one, let me check. I might have missed integrating the fix as I still have the branch open.

viking2010, 2023-08-25:
Cheers dude!