hultenvp/solis-sensor #309: solis_energy_today sometimes goes to zero before the end of the day and come back later again.

pvdw0310, 2023-07-17:
In Home assistant I use Solis_energy_today that I get from the Solis api. 
Sometimes (not every day) it goes to zero before the end of the day and then comes back again.

[image]
If I look then to the Solar production graph, I get a dubbel production that day.
[image]
If I look to the solar detail production for that day I get:
[image]

So the day / week / month / year calculation is total wrong then.
I don't know if this is a problem in the Solis api or from the Solis integration to home assistant.

Please advice what I can do to prevent that problem.
Best regards
Peter

Remppa, 2023-07-17:
I have exactly the same issue. Started to happen Jul 15th. One day after taking solis integration into use.

viking2010, 2023-07-18:
It's been happening for a while and I believe the last update was meant to include some functionality to help alleviate the problem. So far (the last 3-4 days), mine seems to have been ok, but it's sporadic. Some nights it works as it should with no large spikes, and another nights it can be miles out. This has been logged previously on here as an issue. If you check some of the closed issues you'll see similar problems others have had.

Remppa, 2023-07-18:
Good to know. It seems to be happening between hours 23-00. Is this something that the solis inverter is sending/providing or is there some interesting if clause that will produce this behavior. Maybe need to open the plugin into IDE and check if there is some place related to this energy today value.

viking2010, 2023-07-19:
It actually started happening around the time of the clock changes for many, so not sure if that's related. But so far, running v3.4.0 of the integration it's running without any issues. 

What happens is, say for example you generated 20kWh during the day, the Solis API is sends the correct information from 23 to 00, but the integration resets it to zero. This then means when the Solis API is queried again, at which point it adds 20kWh to the previous total. This can happen anywhere from 1 to 5 times (or so I've seen) between the 23 to midnight hours. I think there was an explanation in one of the other posts about this but can't recall what it was. I just edit the statistics when it happens and sometimes if I can be bothered, alter the DB tables at the back end to change the zero values.

viking2010, 2023-07-20:
I spoke to soon! First time in about a week that I had the spike again from 11pm. @hultenvp, are there any debug logs you need to help with this?

Remppa, 2023-07-20:
> I made myself a total solar energy sensor. I made it with device & servises -> helpers -> create helper -> integration - Riemann sum integral , give a name and set the input sensor to Solis AC Output Total Power and check Left Riemann sum . I set the decimals to zero. My sensor is about the same as te solis total day energy sensor and for the moment I didn't had any problems with the faults between 23 and 00 hour. I Have seen that there is a small solar production at midnight. But that is a minor fault than the dubbel day production fault.

I was also considering the workaround with Riemann sum integral, but didn't put my hands on to this yet. Do you have any snippets you would like to share about your workaround here?

pvdw0310, 2023-07-20:
> 
To make the new energie sensor follow the the steps I wrote : 
Go on home assistant to -> device & services -> helpers -> .... then you make the new sensor and that sensor you can use in de solar energie dashboard.

hultenvp, 2023-07-20:
Yeah, I was already afraid the pull request I got was incomplete, because it expects the inverter to be off when it happens.
I'm working on a more robust fix. Trick is that the sensor has to be stateless or at least robust against restarts so no Riemann sums for me ;-)

Remppa, 2023-07-20:
Yeah. Inverter seems to be online many days in a row with good weather here in Western Finland.

hultenvp, 2023-07-30:
I've run a better fix for a while now and it seems robust, so pushing it to release.
Maybe interesting for you @Remppa to keep an eye on it. I now ignore all zero measurements after noon.
As far as I can reason this should also work fine in the middle of the winter in northern latitudes, but best to keep an eye on it.

closed 2023-07-30 by hultenvp

pvdw0310, 2023-07-30:
Thanks for updating the new fix. I'am to updates, so do I have to wait till home assistant let me know that there is a update or can I do it manually. If I have to do it manually, how can I do that?

Many thanks.

Remppa, 2023-08-01:
Thanks for the fix. I'll keep my eyes open for this 👍