hultenvp/solis-sensor #312: Inverter data missing "total" load as ECS/emergency data i not included

exalyte, 2023-08-09:
**Describe the bug**
The HA integration "total inverter load" only includes things on the grid side, its excludes circuits on the ECS/backup/emergency side of the inverter 

**Config**
stock config with no changes at all 

**Versions**
Installed version
2023.7.2
v3.4.1

**Debug logs**
N/A
logger:
N/A

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

[image]

to summarise 
The integration ignores the ECS side of the inverter output resulting in odd behaviour, if the ECS side draws from the grid it shows as its using the grid, but if its 100% solar/battery it is ignored from the total count

Proposed fix
add the ECS "backup load power" as its own entity, change the current "total" to grid, add a new entity which is sum of both 

outside of this much love for this integration its truly awesome! 

extras;
i wouldnt mind battery stats from the BMS in the integration such as voltage BMS load SOH SOC etc just to add more detail to my tracking. 

thanks

Exa

exalyte, 2023-08-09:
an example of how it present as most my load is on the ECS side (home office, kitchen, servers etc) 

[image]

as you can see the battery is kicking out 1.13kw but the "total load" is only 239w (rest of the house) 
also damn i need to trim the load on the rest of the house lol

hultenvp, 2023-08-21:
Hey @exalyte,

Not entirely sure what you mean. 
I expose the following power details (sensor name: API name), see [api doc](https://github.com/hultenvp/solis-sensor/blob/master/doc/SolisCloud%20API%201.2.pdf) for more details.
actualpower: inverterDetail.pac
batpower: inverterDetail.batteryPower
gridpowergridtotalpower: inverterDetail.psum
gridtotalconsumptionpower: inverterDetail.familyLoadPower
All of them are just passed on as-is, I'm not conflating any API details into new sensor data. 

I'm not sure which one you still miss?

Cheers.

exalyte, 2023-08-21:
> Hey @exalyte,
>
> Not entirely sure what you mean. I expose the following power details (sensor name: API name), see [api [...]

Hey hultenvp 

Thanks for the reply and the document and the integration, I think the API call I'm missing is bypassloadpower 
I believe looking over the config and that doc currently you call home and family (assuming my totally inexperienced reading of solisclous_apj.py is correct) 
I tried editing that doc and adding the line in but it didn't change anything 

Sorry I have no idea how to query the API to check it always says unauthorized

I'll rewrite the issue I'm seeing perhaps it'll make more sense. 

I have a hybrid inverter with backup/ecs/emergency output (output 2)
On the backup side I run my home office servers etc so if the grid fails the office stays powered 

The total inverter load from the Integration does not include these numbers 

Example 

Inverter output 1 - 500w
Inverter output 2 (backup) - 1000w 
Solar - 0w 
Battery - 1500w (ignoring losses for ease) 

The integration would show a total inverter load (family/home loadpower) of only 500w the 1000w coming from the backup output (bypassloadpower) isn't included 

This leads to the inverter load presenting as 500w instead of 1500w in home assistant

Hope that makes more sense 👍 

Thanks again!

hultenvp, 2023-08-23:
Het @exalyte,

You can download and run https://github.com/hultenvp/solis-sensor/blob/master/test/apitest_async.py from any PC. If you run it in verbose it will dump all API output as JSON.
There is a link (here)[https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues#testing-issues] describing how to run it.

If we confirm what field to add then I can add it for you.

exalyte, 2023-08-23:
> Het @exalyte,
>
> You can download and run https://github.com/hultenvp/solis-sensor/blob/master/test/apitest_async.py from any PC. If you run it in verbose it will dump all API output as JSON. There is [...]

Hey @hultenvp 

That took me a wee bit of time, had to install python then it was asking for dependancies, then it was shouting about yaml not found, then i read the readme and realised i should read instructions sometimes :) 

So here is the section of the output which i think mentions it theres actually a few times it crops up but this lines up with what im seeing on the inverter screen ive put it in bold as its quick long 
i would also wonder, and id need to test it by forcing a grid power loss event but right at the top you have backuptodayenergy i assume that is 100% offgrid which might be cool to add 

cheers again for this really appreciate your assistance 

[
   {
      "acOutputType":0,
      "backupTodayEnergy":0.0,
      "backupTodayEnergyStr":"kWh",
      "backupTotalEnergy":0.0,
      "backupTotalEnergyStr":"kWh",
      "batteryCapacitySoc":92.0,
      "batteryModel":1,
      "batteryPower":-2.577,
      "batteryPowerPec":"1",
      "batteryPowerStr":"kW",
      "batteryTodayChargeEnergy":9.6,
      "batteryTodayChargeEnergyStr":"kWh",
      "batteryTodayDischargeEnergy":5.4,
      "batteryTodayDischargeEnergyStr":"kWh",
      "batteryTotalChargeEnergy":715.0,
      "batteryTotalChargeEnergyStr":"kWh",
      "batteryTotalDischargeEnergy":829.0,
      "batteryTotalDischargeEnergyStr":"kWh",
      "bypassAcOnoffSet":1.0,
      **"bypassLoadPower":2.598,**
      **"bypassLoadPowerStr":"kW",**
      "collectorId":"redacted",
      "collectorSn":"redacted",
      "currentState":"3",
      "dataTimestamp":"1692808495727",
      "dataTimestampStr":"23/08/2023 17:34:55 (UTC+01:00)",
      "dcInputType":1,
      "dispersionRate":0.0,
      "etoday":30.9,
      "etoday1":30.9,
      "etodayStr":"kWh",
      "etotal":1.977,
      "etotal1":1977.0,
      "etotalStr":"MWh",
      "fisGenerateTime":1685721550000,
      "fisGenerateTimeStr":"02/06/2023 16:59:10 (UTC+01:00)",
      "fisTime":"1685721550207",
      "fisTimeStr":"02/06/2023 16:59:10 (UTC+01:00)",
      "fullHour":5.15,
      "gridPurchasedTodayEnergy":2.8,
      "gridPurchasedTodayEnergyStr":"kWh",
      "gridSellTodayEnergy":1.3,
      "gridSellTodayEnergyStr":"kWh",
      "groupId":",redacted,",
      "id":"redacted",
      "inverterConfig":"1",
      "inverterMeterModel":5,
      "inverterSoftwareVersion":"3d0044",
      "inverterSoftwareVersion2":"080000",
      "isS5":1,
      "ivSupport":0,
      "maxDcBus":0.0,
      "maxDcBusTime":"1692808495727",
      "maxUac":245.7,
      "maxUacTime":"1689428607024",
      "maxUpv":316.7,
      "maxUpvTime":"1689091360719",
      "model":"3105",
      "nationalStandards":"53",
      "offlineLongStr":"--",
      "pac":0.805,
      "pac1":0,
      "pacStr":"kW",
      "parallelAddr":0,
      "parallelBattery":0,
      "parallelOnoff01":0.0,
      "parallelOnoff02":0.0,
      "parallelPhase":0,
      "parallelStatus":0,
      "pow1":391.35,
      "pow10":0.0,
      "pow11":0.0,
      "pow12":0.0,
      "pow13":0.0,
      "pow14":0.0,
      "pow15":0.0,
      "pow16":0.0,
      "pow17":0.0,
      "pow18":0.0,
      "pow19":0.0,
      "pow2":415.2,
      "pow20":0.0,
      "pow21":0.0,
      "pow22":0.0,
      "pow23":0.0,
      "pow24":0.0,
      "pow25":0.0,
      "pow26":0.0,
      "pow27":0.0,
      "pow28":0.0,
      "pow29":0.0,
      "pow3":0.0,
      "pow30":0.0,
      "pow31":0.0,
      "pow32":0.0,
      "pow4":0.0,
      "pow5":0.0,
      "pow6":0.0,
      "pow7":0.0,
      "pow8":0.0,
      "pow9":0.0,
      "power":6.0,
      "powerStr":"kW",
      "productModel":"3105",
      "psum":0.0,
      "psumCal":0.0,
      "psumCalPec":"1",
      "psumCalStr":"kW",
      "psumStr":"kW",
      "rs485ComAddr":"101",
      "simFlowState":-5,
      "sn":"redacted",
      "state":1,
      "stateExceptionFlag":0,
      "stationId":"redacted",
      "stationType":0,
      "tag":"zhichuan",
      "timeZone":1.0,
      "timeZoneName":"(UTC+01:00) 
      "timeZoneStr":"(UTC+01:00)",
      "totalFullHour":329.5,
      "updateShelfBeginTime":1677772800000,
      "updateShelfEndTime":1835625600000,
      "updateShelfEndTimeStr":"03/03/2028",
      "updateShelfTime":"5",
      "userId":"redacted"
   }
]

hultenvp, 2023-08-24:
I created a v3.4.2 release to add the sensor. It's called Backup Load Power similar to the description in the API.
Let me know if it works, feel free to create new feature requests if you need additional data exposed, please do not attach additional requests to this ticket.

exalyte, 2023-08-24:
So Far so good, on 3.4.3 

Ill give it a wee bit of time but so far so good, much appreciated!

closed 2024-01-05 by hultenvp