hultenvp/solis-sensor #315: Can not enter plantid anymore.

canedje, 2023-08-15:
This morning I had no converter information anymore.
It says unavailable.
So I removed my Solis integration and try to reinstall.
I’m using the v2 option with:
Https://home.solarman.cn
I did enter my credentials and try to enter my plant_id.
But HA is changing automatically my last figures of the id.
The last 4 figures are:
8640. ha is changing this in 8700.
So I”m not able to initiate

**Config**

## System Information

version | core-2023.8.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.39
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 3916
Installed Version | 1.32.1
Stage | running
Available Repositories | 1271
Downloaded Repositories | 41

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 48.5 GB
disk_used | 10.6 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), NGINX Home Assistant SSL proxy (3.5.0), Samba share (10.0.2), Terminal & SSH (9.7.1), Z-Wave JS (0.1.87), Samba Backup (5.2.0), Duck DNS (1.15.0), ESPHome (2023.7.1), Node-RED (14.4.5)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 28
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 9 augustus 2023 om 11:55
-- | --
current_recorder_run | 15 augustus 2023 om 17:32
estimated_db_size | 2089.61 MiB
database_engine | mysql
database_version | 10.11.4

</details>

**Versions**
HA Version an deployment
HACS version
Integration version

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

hultenvp, 2023-08-21:
Hey @canedje : Can you add the debug logs please?

canedje, 2023-08-21:
I'm on Holliday at the moment. 
I Will try aftershave that

hultenvp, 2023-08-23:
You'll smell fantastic ;-)

hultenvp, 2023-08-24:
See also #318

canedje, 2023-08-30:
The same under windows

hultenvp, 2024-01-05:
Closing due to inactivity. Reopen if problem persists, but please add debug log if you do

closed 2024-01-05 by hultenvp