hultenvp/solis-sensor #317: Integration does not anymore connect to the server.

heebo1974, 2023-08-18:
Integration haven't been connected to server for couple of days.
I haven't done any changes for a long time, only updated home assistant.
I am using ginglong_v2 server with Solar Sofar inverter.

**Versions**
Home Assistant 2023.8.2
Supervisor 2023.08.1
Operating System 10.4
Frontend 20230802.0 - latest
HACS 1.32.1
Integration version 3.4.1

**Debug logs**
```
2023-08-18 07:24:48.181 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2023-08-18 07:24:48.181 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2023-08-18 07:24:50.554 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'aiohttp.client_exceptions.ContentTypeError'>
2023-08-18 07:24:50.554 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 60 seconds.
2023-08-18 07:24:50.555 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
2023-08-18 07:25:52.754 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'aiohttp.client_exceptions.ContentTypeError'>
2023-08-18 07:25:52.754 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 120 seconds.
2023-08-18 07:25:52.756 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 120 seconds.
```

heebo1974, 2023-08-21:
Here is something more:
[image]

hultenvp, 2023-08-23:
I'm not sure what's wrong in the response given this input.
I recommend you follow the steps described here: https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues#testing-issues
Let's see if that sheds a light on what the issue is. Also check the troubleshooting steps in the readme.

heebo1974, 2023-08-24:
I deleted my settings and tried to create new one. Could it be possible that this legacy server (home.solarman.cn) is not anymore available ? The new site is https://home.solarmanpv.com/, but that did not work either with integration. Well I don't know the port ? I tried https://home.solarmanpv.com:13333, but it did not help. 

Any ideas ?

hultenvp, 2023-08-24:
I guess they indeed shut down the legacy server, though I can still ping the server?
I do not have a solarman myself, so cannot help you, but from what I read [here](https://community.home-assistant.io/t/collect-inverter-data-from-solarman-api/328760) for the new service the API's differ from SolisCloud (at least the authentication already does).
So I guess you'll have to use the steps described in that thread or maybe, if you have an older WiFi dongle, [this](https://github.com/StephanJoubert/home_assistant_solarman)

canedje, 2023-08-30:
I'm using a very old Omnik converter. what is working with Solarman and solis websides.
But I'm not able to connect directly to the wifi unit (I can connect but not read local data). So it looks like I'm stuck :-(

hultenvp, 2023-08-30:
Look at the "here" link in my previous post. It should give you a way to read out solarmanpv.com.

hultenvp, 2024-01-05:
Close as solved