hultenvp/solis-sensor #318: Can't complete integration adding

ChrisMarriott38, 2023-08-18:
I get a strange blank set of fields to input when i try to add this. After i've selected the solis cloud and what i think are the correct credentials it just says Auth. And cant go further? Any ideas
[image]

hultenvp, 2023-08-21:
I'm not sure what's going on here. Looks like missing resources as there should be descriptions with every field. Best to try to remove and reinstall the integration.

ChrisMarriott38, 2023-08-21:
Yea I've tried that sadly a couple of times :( I don't suppose you have any other ideas?

hultenvp, 2023-08-23:
The line number refers to the error output. Your config is wrong.
What bothers me is you do not see strings.
If you manually add the integration, without HACS, then please ensure you also have the translations folder plus the en.json

In any case these are screenshots the names of the individual fields. Make sure you follow the instructions carefully because any config mistake will always result in "No inverters found" as the API does not tell me what's wrong.

[image]

[image]

ChrisMarriott38, 2023-08-24:
I am a bit new to this and for some reason, when i upload to the Synology sometimes it misses a folder. That looks like it was the issue because i uploaded translations folder and the file inside and now the fields populate. 

[image]

One thing i noticed though when this showed up was i was putting things in the wrong order.

On the main page i was reading this part - So i assumed this was the order to enter the info (as i couldn't see the fields). Might be good to idiot proof it to just swap the wording around for the Key ID and Secret - Because i couldn't see the field names i was putting them in the wrong way - But less important now that i can see the field names 👍 

*****
Soliscloud

Provide username, key id, secret and station id. If you want to add multiple plants just repeat "add integration" for each plant.
*****

The info went in first time and it added the Integration flawlessly. And all the data pulled in perfectly! Amazing work! And sorry that my blunder added this thread! Hopefully it helps anyone in the future if they are silly like me and didn't add that critical folder!

hultenvp, 2023-08-24:
Glad it works! Closing the issue.

closed 2023-08-24 by hultenvp