hultenvp/solis-sensor #325: Solis data updates only every 5 minutes while portal updates every minute

gijssenh, 2023-09-05:
Solis has increased the data updates from 5 to 1 minute on my request. I see the change in the Soliscloud portal. There data is updated every minute, but in HA the update frequency stayed at every 5 minutes. I guess I am missing something ... Can anyone help?

snigehere, 2023-09-06:
I have the same - Solis moved my refresh to 1 minute.

Take a look here https://github.com/hultenvp/solis-sensor/discussions/100 
This gives a useful work around but you would need to remember to update each time the code is updated.

The ideal solution would be a parameter that can be changed .. @hultenvp any chance you could look into this?

hultenvp, 2023-09-07:
We could make this a [configuration options](https://developers.home-assistant.io/docs/config_entries_options_flow_handler/). I've made this an enhancement. I do not have much time, so community contributions are welcomed ;-)

hultenvp, 2023-09-07:
See #234

viking2010, 2023-09-08:
I too have mine set to 1 min upload cycles. For a while it kept reverting to 2 mins (worth looking out for that) and I needed to have a beta firmware updated on my wifi dongle which has resolved the issue. I will say that whilst it's set to 1 min intervals, it doesn't always grab them every minute in HA. 90% of the time it does, but there are times that it can be up to 5 mins. I put this down to if the network at their side is busy or a request timed out. 

Rather than allowing the update intervals to be a selectable option/range, the update cycle interval is viewable in the API for the data logger. You could use this value to automatically set the update frequency in the integration. I'm only just getting there with Python, and not really at the level yet to build an integration or even offer up the code to do so, but, I'm sure there are others that can!

snigehere, 2023-09-08:
So having recently switched to one minute cycle I can see that the data is updating more regularly however its stale data... looks like it's taking 4 to 5 minutes to get up and down.

I am looking at the sensor my_solis_inverter_timestamp_measurements_received which i assume is when solis cloud got the data i converted it to real time and its 5 to 6 minutes behind the log timestamp (shown below) .... 

Is this just me? Or are others seeing the same?

last solar update changed to 08-Sep-2023 15:32:51 triggered by state of my_solis_inverter Timestamp Measurements Received changed to 1694183571.0
15:38:28 - 9 seconds ago
last solar update changed to 08-Sep-2023 15:31:36 triggered by state of my_solis_inverter Timestamp Measurements Received changed to 1694183496.0
15:36:17 - 2 minutes ago
last solar update changed to 08-Sep-2023 15:30:23 triggered by state of my_solis_inverter Timestamp Measurements Received changed to 1694183423.0
15:34:12 - 4 minutes ago
last solar update changed to 08-Sep-2023 15:29:10 triggered by state of my_solis_inverter Timestamp Measurements Received changed to 1694183350.0
15:33:10 - 5 minutes ago

snigehere, 2023-09-08:
ok - so maybe its a volume issue because later the difference improved
last solar update changed to 08-Sep-2023 16:09:34 triggered by state of my_solis_inverter Timestamp Measurements Received changed to 1694185774.0
16:10:49 - 3 minutes ago

I assume the data received timestamp comes from the cloud and is associated to a package of data....

luftdieb, 2023-09-13:
Hi together, if you are looking for a quicker update of the results, you should have a look to https://github.com/wills106/homeassistant-solax-modbus
It's updating the results by default every 15 seconds and it's amazing, what you can control by this integration. It's available in HACS. I don't use any other RS485 adapter. Just the Solis LAN adapter, which transfer the data into Solis cloud. But this adapter offer an TCP-Modbus Server on port 8899. You just have to enter the IP of the adapter and the port 8899. That's all! No changes on Adapter Settings required.

snigehere, 2023-09-13:
Thanks for the pointer ... I installed this integration to see if it would work for me .. the install instructions seem a bit scant on details ... I now have to wait for the sun to come up... 

When you say you have the "Solis LAN adapter" do you mean the wifi data logger or some other bit of tech?

viking2010, 2023-09-14:
Hi, thanks for the information. Does that adaptor allow remote control? I used to have the old wifi dongle but that is no longer supported for remote control. That exposed port 8899 too. There is only one other dongle I know of that supports either modbus or solis cloud but not both at the same time, and that's the SOL-S2-WL-ST.

snigehere, 2023-09-14:
Didn't work for me - I tried a few different configuration options but no useful connection ... I am not sure the wifi stick offers any service 
2023-09-14 19:12:19.005 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'name': 'solis_mobus', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solis', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': True, 'host': '192.168.0.249', 'port': 8899, 'tcp_type': 'tcp'}
2023-09-14 19:12:19.005 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-09-14 19:12:19.005 DEBUG (MainThread) [custom_components.solax_modbus] Setup solax_modbus.solis_mobus
2023-09-14 19:12:19.005 DEBUG (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2023-09-14 19:12:19.005 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
2023-09-14 19:12:19.005 DEBUG (MainThread) [custom_components.solax_modbus] solis_mobus: ready to call plugin to determine inverter type
2023-09-14 19:12:19.005 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] solis_mobus: trying to determine inverter type
2023-09-14 19:12:24.030 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis_mobus: reading serial number from address 0x80ec failed; other address may succeed
2023-09-14 19:12:24.030 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read solis_mobus 0x80ec serial number: None, swapped: False
2023-09-14 19:12:24.030 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis_mobus: cannot find serial number, even not for other Inverter
2023-09-14 19:12:24.031 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis_mobus inverter type - serial number : unknown

My guess is that the wifi stick may not have a mobus server

luftdieb, 2023-09-14:
> Thanks for the pointer ... I installed this integration to see if it would work for me .. the install instructions seem a bit scant on details ... I now have to wait for the sun to come up...
> 
> When you say you have the "Solis LAN adapter" do you mean the wifi data logger or some other bit of tech?

> Hi, thanks for the information. Does that adaptor allow remote control? I used to have the old wifi dongle but that is no longer supported for remote control. That exposed port 8899 too. There is [...]

Hi Viking2010,
I'm using the Solis DLS-L LAN stick (https://solis-service.solisinverters.com/en/support/solutions/articles/44002092757-solis-dls-l-lan-stick-troubleshooting). THis stick allow also control parameter in Solis inverter, as you can see on the screenshot 
[image]
[image]

I've tried also many thing in the early beginning one year ago without success reading data directly from TCP-Modbus server. I've lerned by adapter is too new and therefore it's not working. At this time I was really happy to find  the cloud solution provided by hultenvp/solis-sensor. 
But with this solution I've seen yesterday my dreams all become true :-)

I hope you can also manage to connect your HA with the Solis inverter directly by TCP-Modbus.

luftdieb, 2023-09-14:
> Didn't work for me - I tried a few different configuration options but no useful connection ... I am not sure the wifi stick offers any service 2023-09-14 19:12:19.005 DEBUG (MainThread) [...]

Hi snigehere,
you can check if the port 8899 is offering a service by your Wifi Stick or not by a portscan or under Windows with powershell command "test-netconnection 192.168.0.249 -port 8899".
Under linux you can run nmap "sudo nmap -p8899 192.168.3.238".

snigehere, 2023-09-15:
Thanks - it did respond to the test
ComputerName     : 192.168.0.249
RemoteAddress    : 192.168.0.249
RemotePort       : 8899
InterfaceAlias   : Ethernet
SourceAddress    : 192.168.0.43
TcpTestSucceeded : True

So more reading required.....

luftdieb, 2023-09-17:
@snigehere: Maybe you can try to get more support by the git repo. There are more users, asking for support to get their Solis inverter connected. They have been asking for serial number. Maybe there is a chance to find a solution also for your hardware.
I just can tell you, my solis adapter was originally not support (too new), and now it's working. I hope you will find also a solution. 
I've tried at this time several tools / programs and I made some positive experience with some Android TCP-Modbus tools and some bad experience with windows tools. So, with the Android tools, I could fetch some of the parameter. 

Sorry for no better news / help.

snigehere, 2023-09-17:
@luftdieb thanks for your help and pointers.

jmason, 2023-09-18:
> Rather than allowing the update intervals to be a selectable option/range, the update cycle interval is viewable in the API for the data logger. You could use this value to automatically set the [...]

Hey BTW that was the idea behind this PR: https://github.com/hultenvp/solis-sensor/pull/240 -- I think that had to be reverted though, due to some impact here: https://github.com/hultenvp/solis-sensor/issues/254 and I haven't had a chance to dig into why that happened. :(

klotzenkuhle, 2023-09-20:
@luftdieb It works like a charm, thanks a lot.

gijssenh, 2023-09-22:
I "seem" to have solved the problem with the update frequency of my Solis. 
Soliscloud was showing updates every minute but in HA updates were shown only every 5 min.

I changed the following line in /config/custom_components/solis/service.py:

**_# Match up with the default SolisCloud API resolution of 1 minute
SCHEDULE_OK = 1_**

Previously SCHEDULE_OK was initialized on 5 minutes.

Now data in HA is updated synchronous with the data on the Soliscloud website.
Maybe this little change has implications on other points in the software? (I'm familiar with software, but far from a specialist ...)
Can anyone advise?

Thanks

gijssenh, 2023-09-22:
At worst the difference between real data and data shown in HA could now be 2 minutes. At best it is one minute.

snigehere, 2023-09-23:
I think that sensor.my_solis_inverter_timestamp_measurements_received shows the timestamp of the data as it arrived at the solis cloud from the inverter .. if you compare that to the current time when it updates in HA it shows the delay through solis cloud assuming that HA is in sync with the time source.  (nb: I am not sure they store the timestamp it was sent from the inverter as that is prone to error as its user set so there could be a delay inherent on that side as well).

For me the my_solis_inverter_timestamp_measurements_received timestamp is, for the most part, between 1 and 2 minutes delayed but I have seen it 5 to 6 minutes delayed .. my guess is this happens when there is extra load on the cloud and API,

So "at worst" is not my finding .. "at worst" is a longer time at times ... It also really depends on how you use the data to know if its an issue or not .. if you have fine grain automations that are supposed to deal with surplus energy than you will need to look for an alternate as you could be triggering the automation late or have it running long.

hultenvp, 2023-10-08:
@snigehere is correct, you can tweak the update schedule as @gijssenh shows, but you cannot control the latency introduced in the cloud (well, at least this cloud application ;-) )
Local solutions are better in that case.
I still plan to add control over the update schedule via the options. 
Just got to find the time, though.

xion2000, 2023-10-28:
> Solis has increased the data updates from 5 to 1 minute on my request

How can I do this?
I've just checked the Solis Cloud portal and all 3 of my systems are updating every 5 minutes

snigehere, 2023-10-28:
> How can I do this?
> I've just checked the Solis Cloud portal and all 3 of my systems are updating every 5 minutes

Reach our to solis support ( euservice@solisinverters.com ) and ask them to make the change - its a setting on their side

xion2000, 2023-10-28:
> > How can I do this?
> > I've just checked the Solis Cloud portal and all 3 of my systems are updating every 5 minutes
> 
> Reach our to solis support ( [euservice@solisinverters.com](mailto:euservice@solisinverters.com) ) and ask them to make the change - its a setting on their side

Cheers, I've sent them an Email including all my details for each system

xion2000, 2023-10-28:
I got a quick reply...

Hello Graham,
 Actually, 5 mins uploading cycle is mandatory for all end user we can't reduce it to 1 min.

snigehere, 2023-10-28:
I wrote to them about getting server b working on my data stick.. I told them that the 5 minute delay manifests as 10 minute taking into account the cloud delays.  I pointed out that as there was no way to get local data it made it impossible to manage usage.  After a few back and forth mails they told me they had changed me to 1 minute cycle.. I checked the data and it reflected that... I will recheck tomorrow.  If you message me your email I will share the email

ccj2740, 2023-12-07:
> Solis has increased the data updates from 5 to 1 minute on my request. I see the change in the Soliscloud portal. There data is updated every minute, but in HA the update frequency stayed at every 5 minutes. I guess I am missing something ... Can anyone help?

I have a issues with one sensor that still only update every 5min , so made a post here :
https://github.com/hultenvp/solis-sensor/issues/337

If anyone could help I would appreciated it very much.

chronvas, 2024-01-17:
> Hi Viking2010, I'm using the Solis DLS-L LAN stick (https://solis-service.solisinverters.com/en/support/solutions/articles/44002092757-solis-dls-l-lan-stick-troubleshooting). THis stick allow also [...]

I couldn't find this stick for sale anywhere, any tips where to find it?

TheGroundZero, 2024-02-12:
Looks like the WiFi datalogger doesn't expose the Modbus port (nmap shows closed port).

Can I wire a RS485 to the inverter and get it integrated in HA that way?
I believe COM is already occupied to connect with battery (hybrid inverter).

In a distand past, I followed this technique with a RPi Zero and RS485 HAT for a different inverter. But that required me to list all modbus addresses in a big script: https://sequr.be/blog/2021/08/reading-ginlong-solis-inverter-over-serial-and-importing-in-home-assistant-over-mqtt/

[image]

fboundy, 2024-12-07:
This summarises the position abou a year ago. Things have moved on a bit but the basics are the same.