hultenvp/solis-sensor #326: Energy Dashboard doesnt have access to all sensors?

lynxus, 2023-09-08:
Hi, Not sure if this is a duplicate, but cant find much on it.

Running:
Home Assistant 2023.9.0
Supervisor 2023.08.3
Operating System 10.5
Frontend 20230906.1 - latest

Ive installed Solis Sensor, It works fine.
From within the Integration entities I can see the sensors showing correctly.

For example:
[image]

however, when setting up the Energy dashboard, those same sensors are not available.
It looks like about 25% of what is in the integration is only showing.

Any ideas what could be wrong here?

-Graham

lynxus, 2023-09-08:
Further to this, If anyone has any info on how to actually setup the Energy dashboard, that would be great. 
It seems that none of the values in the dashboard match the Cloud.

michaelsargent, 2023-09-13:
Hi

You need to wait a few hours for the energy dashboard to reflect useful information.

And on another note SolisCloud seems to be not updating today. This issue is in SolisCloud and not the integration. My logger is reporting correctly and reports that it is connected.

Regards

Michael Sargent
M: 0421 047 231

hultenvp, 2023-10-07:
Energy dashboard (for electricity) only takes in sensors that use a unit of energy (kWh, MWh etc.), not units of power (W, kW), current or voltage.

closed 2024-01-05 by hultenvp