hultenvp/solis-sensor #329: No labels appearing on either of the 2 setup forms.

tom-cox, 2023-10-08:
**Describe the bug**
On the 2 forms for setting up the integration there are no labels.

Steps to reproduce:
1. Go to Settings > Devices & Services > Add Integration
2. Select Solis
3. Form Appears // The form has no labels to explain the purpose of the 2 fields at first I didn't spot anything was missing but the next form is more significant
4. Select either option (Ginlong / Soliscloud) and click submit
5. Form Appears // This form is much more of a problem there are no fields and no prefilled values apart from the numbers "00" in the password field.

In order to workaround the above I had to inspect elements and work out what the fields were for based on the field IDs but I managed to get it to work this way.

I inspected the code and the labels are in the strings.json file but for some reason are not appearing on the form itself, I tried to work it backwards myself but so far no joy, If I crack it I'll add it to my commit

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment
HACS version
Integration version

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

hultenvp, 2023-10-27:
Hi @tom-cox,

How did you install the integration? Issue looks familiar, see #318

hultenvp, 2024-01-05:
Closing due to inactivity