hultenvp/solis-sensor #331: Inverter Control API

HarryFlatter, 2023-10-14:
Apologies if this is in the wrong place. It's an enhancement request rather than a bug.

It seems that on request and signature of NDA, Solis will now provide access to their Inverter Control API (v2,0).

Is this something that could be included with this or a new integration for Solis inverters? It would make a neat solution and obviate the need to go the MODBUS route.

eddysteurs, 2023-10-22:
I noticed earlier today this solution https://github.com/stevegal/solis_control but it would be much better if this functionality would be integrated in this soliscloud integration

HarryFlatter, 2023-10-23:
That looks exciting - I'll give it a shot, though it looks a bit unstable just at present - CID-wise.

hultenvp, 2023-10-27:
Looks like a dup of #278

Got some discussion there how this conceptually could be added.
Propose we continue discussion there.

closed 2024-01-05 by hultenvp