hultenvp/solis-sensor #337: Issue with delay on "sensor.solis_total_consumption_power"

ccj2740, 2023-12-07:
Hi ,

First I wanna thank you for this great solis-sensor code very good work.
I enjoy it very much and start using for home automations with EV charge as well.

Solis changed my refresh time from 5min to now 1min and I have changed:
SCHEDULE_OK = 1

It works with most sensor and they refresh every 1min, but sensor ""Solis Total Consumption power" seems to still refresh every 5min.
This give me a huge problem as my "Solis Power Grid total power" is updating every 1min and then I get wrong battery status/charge since the load is on "sensor.solis_total_consumption_power" is delayed.

Here you can see the problem:
[image]

Hope some here have a good idea what I could , so it refresh every 1min.
I know alternative is that I buy the Solis DLS-L LAN stick , but then I need to do modbus work and that seems to complicated for me, so I really really hope I can use hultenvp solis-sensor and get refresh every 1min :).

Looking forward for reply :)...

viking2010, 2023-12-08:
Actually, I have now found a change in the SolisCloud_api.py file. In v3.4.3 the "Solis Total Consumption power" (or in the file shown as GRID_TOTAL_CONSUMPTION_POWER is in the Inverter Detail section. In v3.5 this moved to the Plan Detail section. The plant detail section only updates every 5 mins, even if you have 1 min intervals set. This is the case on the website too I believe.

You can see what I mean in this image:
[image]

Not sure if there was a specific reason to move those to Plan Detail or not, but that I believe is why this is happening.

ccj2740, 2023-12-08:
Thank you so much. This is good that I can solve it with downgrade to 3.4.3 version. I will try that during the weekend. Hope maybe "hultenvp" will do a new update later 😀

ccj2740, 2023-12-08:
I have rolled back to 3.4.3 and I can confirm it is updating every 1min again. However sometimes it takes 2min...

viking2010, 2023-12-08:
That's not unusual, especially during the day. Their servers can be under load or network latency can occur  and sometimes a poll is missed. It soon reverts back to 1 min. You may even find that it can take longer than 2 mins at times. Not often, but occasionally it can.

hultenvp, 2024-01-05:
I was offline for quite a while, so memory is not too fresh around this anymore. I cannot find any particular reason why it moved other than that it came in with the PR from @artarus. Guess I overlooked it.
Difference is that it now reports for the whole Plant/Station instead of per inverter, which may  contain multiple inverters if you have any.

@artarus: Any reason why it was moved?

artarus, 2024-01-09:
According to the discussion: "Cant find the sensor for power usage. #327", I couldn't retrieve data from this source 'familyLoadPower' form /inverterDetail, at that time, therefore, another source was used /./stationDetail. I don't know, it might be a server error, but after a long discussion with the Solis, the add some sensor and sensor readings are more or less as in the documentation. I have checked logs today and see "familyLoadPower" in /inverterDetail showing correct values. As for solution a new sensor can be created or moved back to /inverterDetail

ccj2740, 2024-01-15:
Hi @hultenvp ,

I updated to latest version 3.5.1 , but the issue with delay on Solis Total Consumption power is still 5min even if I set it to 1 in your configuration.

I have rolled back to 3.4.3 and I can confirm it is updating every 1min again.

Your latest version have more sensors and I would like to use some of them, so want to ask if you have any plans looking into fixing this in 3.5.x version ? 

[image]

Kind regards
Christian

ccj2740, 2024-01-17:
Thanks @artarus... refer to https://github.com/hultenvp/solis-sensor/discussions/327

I tried the fix.zip and now the "Solis Total Consumption power sensor" is updating correctly again...

BUT the "Solis Total Consumption power sensor" is diffrent data the data i Solis cloud:
image

Example when I compare from Solis cloud to the sensor in home assistant, then I see a difference of around 200-300W...

I think the reason for this difference is that it is taking:
Total Consumption Power - Backup Load Power 2,71 - 0,202 =
image

I don´t think it should include the "backup load power" as it gives me wrong information ?

Also I am Happy to hear that you want to investigate it after 3 weeks I appreciate that very much :)

closed 2024-03-22 by hultenvp

hultenvp, 2024-03-22:
Integrated fix.zip.
Will be released in 3.5.2