hultenvp/solis-sensor #338: Solis AC Output Total Power: unknown when zero (night-time)

hawkeye100, 2023-12-08:
Integration has been working really well for some time - thank you.

More recently (likely since HA update/upgrades), I have noticed that in the evening/overnight the AC Total Power entity is showing as 'unknown', instead of zero.  This causes a display issue on a graph that is expecting numerical data.

The export data from Soliscloud shows numeric zero rather than blank or other value at the same time.

During the day when power is being generated, there is not an issue.

I suspect there is a simple explanation, and/or I have missed something in which case I apologise though I have been unable to find a relevant mention of 'unknown' in another issue here or on HA forum (though could have missed it)

Versions:
Solis sensor (installed from HACS): 3.50
HA:
Core 2023.11.2
Supervisor 2023.11.6
Operating System 11.1
Frontend 20231030.2

My initial thinking as a workaround is to create an automation (or maybe a template sensor) that checks the value and if it is 'unknown' then sets it to zero (untested as yet) and use that where I need it.

The only unusual thing about my HA is that for some reason I have 2 Solis inverters listed, though the other is disabled, as are its entities to avoid conflict (I did not want to delete the second inverter in case it messed up the working config). The other inverter has an inverter serial beginning with many zeroes and ending with some digits from the actual working inverter/config. [wondering if this is legacy from a previous upgrade perhaps]

Here are some screenshots. Any help gratefully received, thank you.

[image]
[image]
[image]

SadGamerGeek, 2023-12-16:
I’m also getting exactly this. 

Versions:
Solis sensor (installed from HACS): 3.50
HA:
Core 2023.11.3
Supervisor 2023.11.6
Operating System 11.1
Frontend 20231030.2

hultenvp, 2024-01-05:
I do not see that, do you perhaps restart HA somewhere during the night?

viking2010, 2024-01-05:
What I find happens is that the value never goes to zero. The last visible value from when there's no more light (usually 1 watt) remains until midnight. Then, the value changes to zero until the morning when it starts generating again. However, if HA is rebooted during the night for whatever reason, the Current PV Power value (solis_inverter_ac_output_total_power) changes to Unknown. It isn't updated again until the morning when power is being generated. The dc_power_pv1/pv2 values go to zero, so I had considered using the two PV array values I have, adding them together in a template sensor and set the solis_inverter_ac_output_total_power to the total of those two values. But it's only 1 watt, it doesn't make any different in the grand scheme of things! lol

But, I suppose it would be good to know why the solis_inverter_ac_output_total_power value isn't being picked up as zero when Solis does show the value on their portal as zero.

SadGamerGeek, 2024-01-05:
Exactly the same here. The DC Power PV1 and PV2 both drop to zero after sunset, but the total output power just seems to stay on the most recent >0 value (generally between 1W & 4W) until midnight when it changes to zero. For me it can go back to a very small value again well before sunrise. No reboots involved.

hawkeye100, 2024-01-05:
> I do not see that, do you perhaps restart HA somewhere during the night?

Not to my knowledge.
It has happened again recently.

This is my workaround (do not pretend is is good or pretty but it seems to work for me)  - a template sensor that I then use in the graphs:
`  - sensor:
      - name: "Solis AC Power correction"
        unit_of_measurement: "W"
        # correct 'unknown' to zero
        state: >
            {% if is_state('sensor.solis_ac_output_total_power', 'unknown') %}
               {{  0 | round(2,default =0) }}
            {% else %}
               {{ states('sensor.solis_ac_output_total_power') | round(2, default=0) }}
            {% endif %}
`
(I put the above in code brackets but not sure it formatted right - hope you can make sense of it)

https://github.com/hultenvp/solis-sensor/issues/338#issuecomment-1879032218:
I get 7W when it gets dark and then zero from midnight (some nights) and have assumed this is from street lights that switch off then, but I may be wrong especially as on others we just get 'unknown'

hastaboera, 2024-01-09:
> What I find happens is that the value never goes to zero. The last visible value from when there's no more light (usually 1 watt) remains until midnight. Then, the value changes to zero until the [...]

Hi, i experience the same issue. Sensor remains with last values untill midnight to only then go to 0.
[image]

I found that the HA sensors last update is before the last update to solis cloud before it goes offline.
[image]
[image]

SadGamerGeek, 2024-02-24:
Not sure if it's related, but since updating to the latest HA Core (2024.2.2 -  though I've just seen there's now a 2.3), I'm getting relatively frequent unknown values even when the sun is out.
[image]

hultenvp, 2024-03-22:
Integrated @hozob's fix.
Will release in 3.5.2

closed 2024-03-22 by hultenvp