hultenvp/solis-sensor #344: Missing 'Charge' & 'Discharge' entities

HarryFlatter, 2024-01-08:
First of all, many thanks for providing the missing entities that I reported in [#334](https://github.com/hultenvp/solis-sensor/issues/334) :-)

I now believe there are 4 battery 'Charge' and 'Discharge' entities missing from the integration.

For 'Charged' the integration provides:

sensor.solis_daily_energy_charged
sensor.solis_total_energy_charged

but nothing for Month or Year

For 'Discharged' the integration provides:

sensor.solis_daily_energy_discharged
sensor.solis_total_energy_discharged

but nothing for Month or Year

The missing figures do seem to be available according to the Solis app & web portal....
[image]

ececconi99, 2024-01-16:
According to the latest Solis API documentation(v2.0) in section 3.4 you can
" 3.4 Obtain real-time data of a single inverter on a certain day
Interface Name Obtain real-time data of a single inverter on a certain day
Interface Description Corresponding daily chart of SolisCloud platform inverter details.
Request URL https://www.soliscloud.com:13333/v1/api/inverterDay
Interface frequency limit 2 times/sec"

however when I try to set up the Solis integration into Home Assistant with that URL (https://www.soliscloud.com:13333/v1/api/inverterDay) it wont accept my credentials. 
i.e. the same credentials which are accepted when I set up the integration using just the basic https://www.soliscloud.com:13333
However this latter set up does not offer the choice of instantaneous power readings. It only offers cumulative daily or monthly figures.

Has anyone else come across this and got a workaround?

HarryFlatter, 2024-03-07:
Hi, any update on these missing entities? Thx

ececconi99, 2024-03-08:
I'm afraid I don't remember in detail what has happened. Going back into my HA i see that my Solis integration is configured with the basic url of https://www.soliscloud.com:13333/ and that it has completed with 1 device and 58 entities. 
I'm using the Sunsynk card in one of my dashboards and selected entities from the above 58 as sources for the display. They are as follows:
Grid: Solis Grid Phase1 Power
Load: Solis Total Consumption Power
Inverter: sensor.sunsynk_grid_power
Battery: Solis Battery Power
Solar: Solis DC Power PV1, Solis DC Power PV2 ( I haven't worked out a way of combining my split solar array so far)

HTH

HarryFlatter, 2024-03-08:
Sorry ececconi99, but you seem to have piggy-backed my original message to hultenvp which was to highlight some possible missing entities around battery charging and discharging, for information that IS available on the app and web portal but not in the integration.

I would appreciate it if you could delete your posts from this thread and raise them in your own query.

Many thanks

viking2010, 2024-03-08:
@HarryFlatter @hultenvp The API needed for this is /v1/api/stationDetail. This info only shows as 0 when using the /v1/api/InverterDetail.

The entity names in the API are: 
        "batteryChargeMonthEnergy": 85.0,
        "batteryChargeMonthEnergyStr": "kWh",
        "batteryChargeYearEnergy": 741.0,
        "batteryChargeYearEnergyStr": "kWh",

Hope this information helps in getting them added to the integration.

HarryFlatter, 2024-03-08:
Many thanks for that @viking2010. Really helpful. 

Not being a 'coder', presumably this would be a change for @hultenvp to make?

Would this also go for the missing 'Discharge' entities too?
eg.
"batteryDischargeMonthEnergy": 85.0,
"batteryDischargeMonthEnergyStr": "kWh",
"batteryDischargeYearEnergy": 741.0,
"batteryDischargeYearEnergyStr": "kWh",

viking2010, 2024-03-08:
Whoops, I forgot to put those in!! But yes, absolutely right. To confirm, they are:

"batteryDischargeMonthEnergy": 82.0,
"batteryDischargeMonthEnergyStr": "kWh",
"batteryDischargeYearEnergy": 758.0,
"batteryDischargeYearEnergyStr": "kWh",

hultenvp, 2024-03-22:
@HarryFlatter: I only see the requested attributes in inverterDetail and inverterDetailList (API v2.0). Not in plantDetail
However, if @viking2010's statement is correct then adding them will have no effect
(note: I don't have a hybrid system, so cannot test battery values myself)

@viking2010: Can you check if inverterDetail**List** gives the correct values?

viking2010, 2024-03-22:
@hultenvp Hi, I can confirm that inverterDetailList does show the battery stats being requested above, but the entity name is different to the ones listed in my previous posts. They are:

"batteryMonthChargeEnergy": 230.0,
"batteryMonthChargeEnergyStr": "kWh"
"batteryMonthDischargeEnergy": 236.0
"batteryMonthDischargeEnergyStr": "kWh"

"batteryYearChargeEnergy": 886.0
"batteryYearChargeEnergyStr": "kWh"
"batteryYearDischargeEnergy": 912.0
"batteryYearDischargeEnergyStr": "kWh"

There's even details for the charge/discharge from "yesterday":
"batteryYesterdayChargeEnergy": 10.3
"batteryYesterdayChargeEnergyStr": "kWh"
"batteryYesterdayDischargeEnergy": 12.1
"batteryYesterdayDischargeEnergyStr": "kWh"

It's interesting that the "Yesterday" figures are to 1 decimal place, yet the current day's entities are whole numbers only, rounded up! Wish Solis would sort that out!

Hope that helps you out @hultenvp! More that willing to do some testing when you need it!

patduckless, 2024-12-13:
I have opened a PR to add these sensors.
https://github.com/hultenvp/solis-sensor/pull/407

HarryFlatter, 2024-12-23:
Hi @patduckless @hultenvp,

Not sure is it's just me, but the yearly figures for both Charge & Discharge appear to be coming through as MWh despite being specced as kWh.

I have posted this on #407 as well. Sorry to duplicate but wasn't sure what to do.

[image]