hultenvp/solis-sensor #349: Question to entities meaning and values

grambas, 2024-01-31:
What is the difference between these two entities:

- Solis AC Output Total Power
- Solis Total Consumption power

Solis AC Output Total Power is mostly in range of  10W~30W bigger. 

I also want to see how much power is given to grid.

- Solis Grid Phase1 Power
- Solis Grid Phase2 Power
- Solis Grid Phase3 Power

entities returns always 0W. Is this bug?

TheGroundZero, 2024-02-12:
`Solis AC Output Total Power` is your total PV production
`Solis Total Consumption power` is your house's total consumption (pv production + grid import + battery discharging - grid export - battery charging)

hultenvp, 2024-03-22:
Hi @grambas 

Adding to @TheGroundZero's answer:
> I also want to see how much power is given to grid.
> 
> * Solis Grid Phase1 Power
> * Solis Grid Phase2 Power
> * Solis Grid Phase3 Power
>
> entities returns always 0W. Is this bug?
No it's not a bug, it means your inverter does not support this. Only hybrid inverters with a battery support these fields and I'm not even sure all hybrid inverters do.

closed 2024-03-22 by hultenvp