hultenvp/solis-sensor #350: no inverters found (since around 9:30am CET

eddysteurs, 2024-02-06:
**Describe the bug**
in soliscloud and solis app updates are still ok (every 5 min) but since a few hours the integration is not receiving the inverters via the api (reload integration and restart HA does not solve the problem)

[image]
**Config**

**Versions**
HA Version an deployment : 2024.1.6
HACS version: latest
Integration version : latest

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

snigehere, 2024-02-06:
Same issue for me

RutgerBeyen, 2024-02-06:
Same here. See also https://github.com/hultenvp/solis-sensor/issues/348

snigehere, 2024-02-06:
I have opened a question ticket with Solis service centre to see if its a generalised issue

Last night I upgraded OS and supervisor to the latest releases

dpgh947, 2024-02-06:
same here, I got this with debug on (replaced anything that looked sensitive with xxxxxx)

```
2024-02-06 10:15:11.741 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inverterList) : ContentTypeError(RequestInfo(url=URL('https://www.soliscloud.com:13333/v1/api/inverterList'), method='POST', headers=<CIMultiDictProxy('Host': 'www.soliscloud.com:13333', 'User-Agent': 'HomeAssistant/2024.1.6 aiohttp/3.9.3 Python/3.11', 'Content-MD5': 'xxxxxxxxx', 'Content-Type': 'application/json', 'Date': 'Tue, 06 Feb 2024 10:15:11 GMT', 'Authorization': 'API xxxxxxxxxx', 'Accept': '*/*', 'Accept-Encoding': 'gzip, deflate, br', 'Cookie': 'aliyungf_tc=xxxxxxx', 'Content-Length': '35')>, real_url=URL('https://www.soliscloud.com:13333/v1/api/inverterList')), (), message='Attempt to decode JSON with unexpected mimetype: text/html', headers=<CIMultiDictProxy('Date': 'Tue, 06 Feb 2024 10:15:11 GMT', 'Content-Type': 'text/html', 'Content-Length': '166', 'Connection': 'keep-alive', 'Via': 'HTTP/1.1 SLB.24')>)
2024-02-06 10:15:11.742 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2024-02-06 10:15:11.742 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 180 seconds.
```

snigehere, 2024-02-06:
All working again for me

dpgh947, 2024-02-06:
@snigehere  working again here too for the past 10 mins or so... maybe your ticket did the job!

RutgerBeyen, 2024-02-06:
Still broken for me though. Ticket with Solis open as well...

viking2010, 2024-02-06:
All seems to be working again for me. How is it for others?

eddysteurs, 2024-02-06:
seems ok since a few hours, will close the case

closed 2024-02-06 by eddysteurs