hultenvp/solis-sensor #352: Sensor holds last reading once inverter is offline instead of going to 0

TheGroundZero, 2024-02-12:
**Describe the bug**

Lately I've noticed the integration would still report e.g. 100W of solar output even though it's dark outside and the inverter has shutdown.
At 01:00 (UTC+1) it appears to drop to 0, so I assume this integration resets the sensor at midnight?
This results in incorrect statistics in HA, as it'll assume I've been producing 100W for the first half of the night. Which quickly adds up to 500-700Wh of "ghost production" every day.

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment: 2024.1.6 - HA Container
HACS version: 1.34.0
Integration version: 3.5.1

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**

[image]

Serafigos, 2024-02-12:
You need to manually create an new entity which sets all values under a certain threshold to 0.

Here my example:

`      - name: "Solis Dach bereinigt" # für Leistung unter 25W wird der Wert auf 0 gesetzt da "Schattenleistung"
        unit_of_measurement: "W"
        state: "{% if states('sensor.solis_ac_output_total_power') is not none %}
              {% set pv_watts = states('sensor.solis_ac_output_total_power') | int(default=0) %}
                {% if pv_watts < 25 %}
                  {{ 0 }}
                {% else %}
                  {{ pv_watts | int }}
                {% endif %}
            {% else %}
              {{ 0 }}
            {% endif %}"`

TheGroundZero, 2024-02-12:
That might be a workable workaround, but ideally the integration would return 0 once the inverter is offline.

Also, the value doesn't drop to e.g. 25W.
At the last sync of the integration, the inverter may still be online and reporting e.g. 111W of production. At the next sync, the inverter is offline, but instead of reporting 0 the integration keeps reporting 111W.
As such, your template will never respond.

Serafigos, 2024-02-12:
In my case 25W works as my inverters are much smaller (1kW) and I assume that your inverter needs at least the 111W to work at all - of course you have to adapt the code (threshold) for your needs. 
I did this workaround a while ago and at that time the solis portal also showed this kind of behavior. Had incoming power up until midnight and then it stopped. I think that the solis database keeps on writing the last known value up until the next comes in.
I assume that this ist still the case in the backend when you access their database through the API. They simply don't show the values in their frontend and also have some kind workarounds like mine at their side.
Hence I still think this is not the integration but a problem on solis side (which they have a bunch of).

hozob, 2024-03-09:
I have made a pull request which should fix this issue: https://github.com/hultenvp/solis-sensor/pull/355

viking2010, 2024-03-19:
> I have made a pull request which should fix this issue: #355

Hi,  just to say I've tested the pull request and it's working well for me! Thank you for this!

hultenvp, 2024-03-22:
integrating PR

closed 2024-03-22 by hultenvp