hultenvp/solis-sensor #353: Issue with sensor.solis_daily_on_grid_energy.

ccj2740, 2024-02-17:
Anyone else be having issues with the :
sensor.solis_daily_on_grid_energy

For me it is started to show 100.9 kWh to grid , but it should only be 10.9 kWh to grid.
[image]

I don´t think it have been doing that all the time 2-3 weeks it was ok is it only me and if so does anyone know if i can change the . from "100.9 to 10.9" ?

hultenvp, 2024-03-22:
@ccj2740 please check your workarounds.yaml in ```custom_components/solis```
Set ```correct_daily_on_grid_energy_enabled: false``` to let the integration stop correcting for an API bug.
Let me know if this fixes it as the correction is by default switched off, so if you didn't change anything then it also shouldn't correct.

ccj2740, 2024-03-22:
@hultenvp I upgraded to 3.5.2 and it solved the problem.
 Thank you so much!!!...

closed 2024-03-22 by ccj2740