hultenvp/solis-sensor #354: Metrics appear to be whole numbers only

deligatedgeek, 2024-03-05:
Not sure this is a bug or just a change in response from Solis API, but since about 1:30am on 4th March only whole numbers are showing in home assistant.  See the change on the graph below
[image]

This is making the energy dashboard look very gappy if it takes 3 hours to use 1KW.
[image]

viking2010, 2024-03-05:
I have the same issue too and can concur that the same timings and issues. I'm running version 3.5.1 of the integration. The same issue is happening with the **Solis Inverter Daily Energy Charged** too. 

[image]

I have checked the online Solis portal for the same period of 4th March and the graphs there are still showing numbers to one decimal place. It's only HA that isn't.

viking2010, 2024-03-05:
Having just run the Solis Cloud API Python script, it's coming out of Solis directly with whole numbers, so this isn't an issue with the integration really, as it only displays what it gets from Solis via the API. I suspect they've altered something on their end which is causing this issue. I'll get a ticket raised with them about it. Once I have a ref number, I'll post it here as it might be worth a few of us mailing them about it to make them realise there's an issue!!

viking2010, 2024-03-05:
Just a bit more info:

When I ran the python script to test the API values returned, I was looking at the values returned from the /vi/api/inverterDetail.

If that is changed to get the data from /v1/api/inveterList (the spelling is correct and not a typo!) then you get the decimals. But I think the integration only looks at inverterDetail, stationDetail or userStationList.

haraldbootz, 2024-03-07:
>. I'll get a ticket raised with them about it. Once I have a ref number, I'll post it here as it might be worth a few of us mailing them about it to make them realise there's an issue!!
> UPDATE: The ticket reference is 333948

email them? to what adress?
or write them via the Solis Service Center?

viking2010, 2024-03-07:
Hi

You can raise a ticket with them via the service portal here: https://solis-service.solisinverters.com/support/tickets

Or you can email them at euservice@solisinverters.com

They have said to me they're investigating, but not heard anything back as yet.

viking2010, 2024-03-07:
I have looked back over the API documentation, and did notice that although there is a reference for /vi/api/inverterDetail that the integration uses, there is also one with a typo called /vi/api/inveterDetail (note the missing R in Inverter).  By changing line 48 in the file soliscloud_api.py from **INVERTER_DETAIL = '/v1/api/inverterDetail'** to **INVERTER_DETAIL = '/v1/api/inveterDetail'** that should bring back the stats showing to 1 decimal place. 

I have informed Solis of this update so they are aware.

deligatedgeek, 2024-03-07:
I have made the change to soliscloud_api.py as requested above, but it has not fixed the issue for me.  I bind mount the home assistant directory and opened a shell in the container to check if the change is there and it is.

I have raised a ticket with solis for the issue, ref 335044 

How do you test the api directly?

viking2010, 2024-03-07:
@deligatedgeek out of curiosity, after making the change, did you restart HA? If not, it will need a restart before it kicks in to effect. As for testing the API directly, there's a python script you can use which I think might be on one of the discussions on this site. 

You will need the latest version of Python installed, plus installing a few modules....think the only one you need is "requests" which you install using "pip install requests" from the python\scripts folder after install (we're talking about a Windows OS install here).

This is the script you'd need to copy/paste into a new file. After installing Python, you can open IDLE to paste the script below into. Then hit F5 to run it. You'll need to enter your API details in the KeyID and secretKey sections to get it to work. You also need to edit the "Body=" section. You need to enter the ID of your device (which can be found at the end of the URL for the device here: https://www.soliscloud.com/#/station/inverter/inverterdetail?id=_This bit is the ID number you need_)  and you also need the Inverter serial number which will be listed on the Device page in the portal.

There are various other API's you can access but some require different details to be entered.

```
import hashlib
from hashlib import sha1
import hmac
import base64
from datetime import datetime
from datetime import timezone
import requests
import json

KeyId = "Enter your Key ID here from the API info you have"
secretKey = b'XXXXXXXXXXXXXXXXXXXX' #Enter your Secret Key here between the quotes

VERB="POST"
now = datetime.now(timezone.utc)
Date = now.strftime("%a, %d %b %Y %H:%M:%S GMT")

Body='{"id":"XXXXXXXXXXXXXXXXX","sn":"160F5221B240033"}' 

Content_MD5 = base64.b64encode(hashlib.md5(Body.encode('utf-8')).digest()).decode('utf-8')
Content_Type = "application/json"

CanonicalizedResource = "/v1/api/inverterDetail"

encryptStr = (VERB + "\n"
    + Content_MD5 + "\n"
    + Content_Type + "\n"
    + Date + "\n"
    + CanonicalizedResource)

h = hmac.new(secretKey, msg=encryptStr.encode('utf-8'), digestmod=hashlib.sha1)

Sign = base64.b64encode(h.digest())

Authorization = "API " + KeyId + ":" + Sign.decode('utf-8')

requestStr = (VERB + " " + CanonicalizedResource + "\n"
    + "Content-MD5: " + Content_MD5 + "\n"
    + "Content-Type: " + Content_Type + "\n"
    + "Date: " + Date + "\n"
    + "Authorization: "+ Authorization + "\n"
    + "Body：" + Body)

header = { "Content-MD5":Content_MD5,
            "Content-Type":Content_Type,
            "Date":Date,
            "Authorization":Authorization
            }
header1 = { "Content-Type":Content_Type,
            "Date":Date,
            "Authorization":Authorization
            }

print (requestStr)

url = 'https://www.soliscloud.com:13333'
req = url + CanonicalizedResource
x = requests.post(req, data=Body, headers=header)
print ("")
print(json.dumps(x.json(),indent=4, sort_keys=True))
```

deligatedgeek, 2024-03-07:
Thanks @viking2010.  
Im running Ha as a container in a docker swarm with the HA directory bind mounted to a local folder. 
I did restart the container, as confirmed the script has the changes.
I may copy your test script to the HA container and run from there, it should already have the correct libraries, if not I'll create virtual env for it.

deligatedgeek, 2024-03-08:
@viking2010 The script worked running inside the HA containers python3 REPL.

Also, And I find this very strange, restarting the container did not introduce the edited code, but restarting HA did, and now I have 1 decimal place again.

Thanks loads for your help

viking2010, 2024-03-08:
That's odd? I would have thought restarting the container would have done that too, Oh well, every day's a school day!! Glad it's working for you again. The only caveat here is that if there's an update to the integration, the change will be reverted, so just look out for that. This is only a temporary sticking plaster to fix the issue! :-) Hopefully it's something @hultenvp can investigate and code in accordingly. Meanwhile, Solis have said that they investigating this and to be patient whilst they do so.

boof-boof-boof, 2024-03-11:
I've updated line 48 in soliscloud_api.py and restarted HA, i seem to be getting correct battery charge/discharge values back now with the decimal place.

I've got a ticket logged with Solis support and have also let them know about this.

boof-boof-boof, 2024-03-14:
I've been told this switch to whole numbers is expected...... which is not a good response to be honest, essentially making the battery metrics useless for the HA energy dashboard and similar reporting. It'd be good if as many people as possible can question this to try and get some traction going on a real fix?

Responses from support:

_"We have just received an update from the cloud team. They have made some changes. They are now indicating that you will receive readings in whole numbers instead of fractions. They found that fractional values were occasionally incorrect, which is why they adjusted the logic to only consider rounded values."

"Our cloud team caters to a diverse range of customers and has identified occasional discrepancies with fractional values. We kindly request that you utilize whole numbers instead. "_

viking2010, 2024-03-14:
Yeah, I've just had the exact same response. It begs the question whether they intend to go to whole numbers across the board in future? That surely doesn't make sense. Also, why would you change something when it's only "occasionally incorrect"? If it was a major issue maybe, but sounds more like a sticking plaster than a fix for an issue that they have.

boof-boof-boof, 2024-03-15:
Yep - it's a pretty bad response. I'm still using the misspelled api which does give decimal metrics and they are perfectly accurate for me judging by solar generation and house consumption so I guess I'll stick with this for as long as possible....

Hopefully enough people complain to Solis to make them put some actual effort in.

viking2010, 2024-03-17:
Solis have also now closed off my ticket. It's disappointing really. I did respond back to them saying if I charge up my battery from the grid by 3.5kWh, then it will now tell me I've used 4kWh which is just completely wrong.  If the stats are occasionally incorrect as a fraction, I'd rather that than being completely wrong as a whole number.  I also said that other areas of the site provide fractions of a number so why not this API?  Solis support did agree with me, and have passed my views on to their API team, but then as I say, closed off the ticket. Whether this will ever change or not, I don't know.

@hultenvp do you think it would be possible to either A) Put a workaround in place to get **batteryTodayChargeEnergy** and **batteryTodayDischargeEnergy** from /v1/api/inverterList rather than /vi/api/inverterDetail? or B) Code a fix to pull the data from /vi/api/inveterDetail (API with the spelling typo) instead of the one with the correct spelling?

For now, the only workaround is editing line 48 in the file soliscloud_api.py to the typo API reference. This will however get overridden with any update.

ccj2740, 2024-03-19:
> whole numbers are showing in home assistant. See the change on the graph below
Hi Viking,

I am not sure if my issue with sensor.solis_daily_on_grid_energy , that now have moved 1 decimal is the same issue as you speak about here ?

https://github.com/hultenvp/solis-sensor/issues/353

hultenvp, 2024-03-22:
Hey all, @viking2010,

I'm uncertain why it's only changed in inverterDetail and not in inverterList or inverterDetailList, which both, according to the 2.0 spec should give back the value for batteryTodayDischargeEnergy in 3 decimals precision.
If they really have trouble with the accuracy then they'd also change the spec for those endpoints.
They probably just have a problem with this integration.

So yes, I can change the code to use inverterList or inverterDetailList, but I'm not sure how long it will hold.
This for sure takes the fun out of maintaining this integration.

viking2010, 2024-03-22:
Hi, yeah it's a pain in the backside for sure. I asked them the same question about how long will it be before you change everything to whole numbers! The response was silence on that one! It does make me think is it worth ditching the Solis portal API route completely and going down a modbus route, but that's a lot more effort and takes time I don't have right now. 

I wasn't sure if coding something into the workarounds file to use InverterList or even using the typo API of InveterDetail was an option. For now, I've just changed to the typo API in Soliscloud_api.py which is working for me and a few others it seems. 

Whilst Solis may be taking the fun out of maintaining this integration for you, just know that your time and effort doing so are very much appreciated!

closed 2024-03-22 by hultenvp

viking2010, 2024-03-23:
> Just released v3.5.2 using inverterDetailList. Let me know if it works

I've just upgraded and sadly, inverterDetailList only shows in whole numbers too. The only place I can find with numbers to a decimal place is /v1/api/inverterList.

blackspiritus, 2024-03-25:
Sadly enough the update didn't fix the issue, still having whole numbers. But the fix with the spelling typo API isn't working anymore too with this version. I reverted to the previous version because of this and it works again with the spelling typo API.
Nonetheless I greatly appreciate everyone helping to get this fixed and the time @hultenvp is putting in this project!