hultenvp/solis-sensor #360: Phase specific measurements missing

jseppeli, 2024-03-24:
Updated to latest version and now missing all the Solis grid phase measurements.
[image]

jseppeli, 2024-03-25:
Tried to reconfigure the integration by deleting inverter and adding it back but no change.

VyckaLTU, 2024-03-26:
Same is for me. I think it has problem with version 3.5.2.  I reinstalled version 3.5.1 and everything works.

jseppeli, 2024-03-26:
I also now did the same.
3.5.1 seems to work without issues.

hultenvp, 2024-03-31:
Appears a side effect of moving to InverterDetailList (to fix #354).
So daily energy charged/discharged doesn't work properly in InverterDetail and the Solis grid phase measurements apparently do not work in InverterDetailList...
Let me see if I can do something about it and make both groups happy. For now, best indeed to stay on 3.5.1.

jseppeli, 2024-03-31:
I'll stick to 3.5.1 then :)
Appreciate the time you put into this @hultenvp!

gintsmurans, 2024-07-16:
Noticed that some measurements are wrong, and found this. For me also phase measurements are not showing and Daily grid energy purchased is wrong. It should have been 10kWh, and entity shows 10, but energy dashboard somehow sums it up to 22kWh.

[image]

DenisTimofijuk, 2024-12-13:
Confirmation that this problem still exists with version 3.7.1. I have created separate issues for a specific variable I am interested in.