hultenvp/solis-sensor #361: Remaining Battery Capacity stays at 100

jeroenbal, 2024-03-27:
Since 3.5.2, the remaining battery capacity stays at 100%

sensor.my_solis_inverter_remaining_battery_capacity

edbanger, 2024-03-27:
I have the same issue, but this doesn't seem related to solis-sensor:
- Issues started after scheduled Soliscloud maintenance at 05:00 UTC on 26/03/2023.
- Issue exists in Soliscloud as well
- BMS and Inverter correctly report actual SOC

I've logged a ticket with Solis: https://solis-service.solisinverters.com/en/support/home

jeroenbal, 2024-03-27:
Oh, thanks for your info. I upgraded the HA solis-sensor that same day...

jeroenbal, 2024-03-28:
It looks like they fixed it last night :-)

closed 2024-03-28 by jeroenbal

edbanger, 2024-03-28:
For me this was fixed overnight in Soliscloud, so the SOC is now correctly reported in solis-sensor as well.