hultenvp/solis-sensor #367: Whole number issue back after update to 3.5.2

deligatedgeek, 2024-04-17:
Probably another Solis API "update", but after upgrading to 3.5.2 only whole numbers are showing in home assistant. See the change on the graph below.
[image]

Same effect as previous Solis change
[image]

**Versions**
HA Version: Docker 2024.4
HACS version: 1.34.0
Integration version: 3.5.2

deligatedgeek, 2024-05-02:
I think the code should be reverted and the misspelled inveterDetail endpoint should be used.  As that endpoint is documented [here]( https://ginlong-it-test.oss-eu-central-1.aliyuncs.com/templet/2022.5.3%20SolisCloud%20Monitoring%20API%201.1.1.pdf)

hultenvp, 2024-08-29:
Hi @deligatedgeek,

Picking up again. Is this problem still there or did they fix the resolution in the meantime?
If it's still there then I need to think hard how to proceed as there are some conflicting issues and I do not want to go into cherrypicking data from various endpoints trying to make right what's made wrong by Ginlong Solis.
(see https://github.com/hultenvp/solis-sensor/issues/360#issuecomment-2028799757)

jmccomb1979, 2024-09-12:
I think this was 'fixed' by Ginlong Solis, although I've noticed over the past few days, reported battery charging/discharging is back to whole numbers, with the fraction left over getting caught up in the 23:00-00:00 period.  This might be due to the Home Assistant 2024.9 update, as graphs before 4th September do not show the whole number jumps..

[image]
[image]

jmccomb1979, 2024-09-30:
After running the soliscloud_test.py it seems that the api is returning whole numbers for InverterDetailList.  JSON output attached for InverterDetailList and InverterDetail endpoints.
[datatest1.json](https://github.com/user-attachments/files/17192041/datatest1.json)
[datatest2.json](https://github.com/user-attachments/files/17192042/datatest2.json)

jmccomb1979, 2024-10-01:
Have opened a support ticket for the battery charge/discharge whole number issue, and Solis re-enabled the API access that I already had.....

jmccomb1979, 2024-10-08:
I got a reply - 

"_We have received the reply as the inverter DetailList is calculated by the SC platform, and inverterList is reported by the inverter, so the two values are different, and the R&D staff recommends using inverterDetailList._"

I have a fix being tested which pulls *only* the battery charge/discharge data from inverterList, but the big fiery ball is not visible in the sky today so I'm not seeing any battery activity atm.  If it works, I'll submit a PR.

leroyboerefijn, 2024-11-04:
@jmccomb1979 hope you've seen some sunlight since! Any luck fixing this issue? :)

patduckless, 2024-12-16:
I have looked at this bug and I am unable to replicate it. 
Can you confirm if this is still an issue for you?
It seems that for my station I recieve a float for both the inverterDetail and InverterDetailList.
[inverterDetail.json](https://github.com/user-attachments/files/18149638/inverterDetail.json)
[inverterDetailList.json](https://github.com/user-attachments/files/18149639/inverterDetailList.json)

jmccomb1979, 2024-12-16:
It's no longer an issue, I've moved to obtaining the data directly from the Modbus connection that the data loggers use.

blackspiritus, 2025-01-08:
For me it's still an issue. I updated again from 3.5.1 to 3.8.1 hoping it would be solved but to no avail.

hultenvp, 2025-02-08:
Should be fixed in #420 based on data retrieved from other users

closed 2025-02-08 by hultenvp