hultenvp/solis-sensor #370: No inverters found

Dino0005, 2024-05-03:
**Describe the bug**
No inverters found

HA Version 2024.5.0 on Home Assistant Yellow
HACS Version: None (manual installation of Solis integration)
Integration version: 3.5.2

Log:
Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: custom_components.solis.soliscloud_api
Fonte: custom_components/solis/soliscloud_api.py:239
Integrazione: Solis Inverter (documentazione, problemi)
Prima occorrenza: 17:22:57 (3 occorrenze)
Ultima registrazione: 17:29:43

No inverters found

[image]

hultenvp, 2024-08-28:
Going through old tickets.

Is this issue still occurring? Please follow the [wiki](https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues) to test if the API works for you.

Dino0005, 2024-08-28:
No, I haven't had this issue again since the day of this report

hultenvp, 2024-09-06:
Thanks, closing ticket.

closed 2024-09-06 by hultenvp