hultenvp/solis-sensor #375: HA inegration of Solis Inverter

chaosnature, 2024-05-28:
**Describe the bug**
HA integration of Solis Inverter Stopped working since the last week and a half. it was working and i could set and test using pscript in Development mode and see it make the changes on the Solis Charge slot.
but now it stopped.  i reinstalled the integration and also disabled and reenabled the integration and still to no avail.

any ideas anyone?

**Config**
```
service: pyscript.solis_control
data: 
  days:
    - chargeCurrent: "25"
      dischargeCurrent: "20"
      chargeStartTime: "00:00"
      chargeEndTime: "00:00"
      dischargeStartTime: "00:00"
      dischargeEndTime: "00:00"
    - chargeCurrent: "25"
      dischargeCurrent: "20"
      chargeStartTime: "00:00"
      chargeEndTime: "00:00"
      dischargeStartTime: "00:00"
      dischargeEndTime: "00:00"
    - chargeCurrent: "25"
      dischargeCurrent: "20"
      chargeStartTime: "00:00"
      chargeEndTime: "00:00"
      dischargeStartTime: "00:00"
      dischargeEndTime: "00:00"
  config:
    secret: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    key_id: "xxxxxxxxxxxxxxxxxxxxxx"
    username: cxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxx
    plantId: "xxxxxxxxxxxxxxxxxxxx"
```

**Versions**
Core
2024.5.4
Supervisor
2024.05.1
Operating System
12.3
Frontend
20240501.1

**Debug logs**
2024-05-28 03:18:23.235 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 160F42218140050
2024-05-28 03:18:23.243 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 160F42218140050
2024-05-28 03:18:25.688 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetailList for energy_today
2024-05-28 03:18:25.692 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:03:57.541169, at 2024-05-28 02:22:23.233068+
```

```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

graemejross, 2024-05-28:
There is a new version of the python script that deals with trailing commas in the json response.  Here's the resolved issue for solis control:  https://github.com/stevegal/solis_control/issues/21

giac96, 2024-07-05:
Integration not working since 4 days. I entered the portal and it is working correctly. 

[image]

hultenvp, 2024-08-27:
The log files show no issue. 
Is the problem still occurring?

giac96, 2024-08-27:
> The log files show no issue. Is the problem still occurring?

no log, I also tried different inverters type, with no solution

hultenvp, 2024-08-28:
I overlooked in your ticket, but this is about solis control, which is not part of this integration.

closed 2024-09-06 by hultenvp