hultenvp/solis-sensor #377: Detected blocking call to open inside the event loop by custom integration solis

Dino0005, 2024-06-05:
**Describe the bug**
Detected blocking call to open inside the event loop by custom integration solis

**Versions**
HA Version 2024.6.0 on Home Assistant Yellow
HACS Version: None (manual installation of Solis integration)
Integration version: 3.5.2

**Logs:**
2024-06-05 21:28:47.872 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'solis' at custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file: (offender: /config/custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:), please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Traceback (most recent call last):
File "<frozen runpy>", line 198, in _run_module_as_main
File "<frozen runpy>", line 88, in _run_code
File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
sys.exit(main())
File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
exit_code = runner.run(runtime_conf)
File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
return loop.run_until_complete(setup_and_run_hass(runtime_config))
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
self.run_forever()
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
self._run_once()
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
handle._run()
File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
self._context.run(self._callback, *self._args)
File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component
result = await _async_setup_component(hass, domain, config)
File "/usr/src/homeassistant/homeassistant/setup.py", line 447, in _async_setup_component
await asyncio.gather(
File "/usr/src/homeassistant/homeassistant/setup.py", line 449, in <genexpr>
create_eager_task(
File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task
return Task(coro, loop=loop, name=name, eager_start=True)
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
await self.async_setup(hass, integration=integration)
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
result = await component.async_setup_entry(hass, self)
File "/config/custom_components/solis/__init__.py", line 82, in async_setup_entry
portal_config = SoliscloudConfig(
File "/config/custom_components/solis/soliscloud_api.py", line 185, in __init__
with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:

**Screenshots**
[image]

alanglund, 2024-06-06:
Logger: homeassistant.helpers.typing
Source: helpers/deprecation.py:197
First occurred: 09:43:46 (2 occurrences)
Last logged: 09:43:46

HomeAssistantType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'solis' custom integration
ServiceCallType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.ServiceCall instead, please report it to the author of the 'solis' custom integration

Dino0005, 2024-06-06:
> Not sure if this is related to the above, apologies I am not an expert in this area.
> 
> Logger: homeassistant.helpers.typing Source: helpers/deprecation.py:197 First occurred: 09:43:46 (2 occurrences) Last logged: 09:43:46
> 
> HomeAssistantType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'solis' custom integration ServiceCallType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.ServiceCall instead, please report it to the author of the 'solis' custom integration

No, that's another issue... Try looking here  https://github.com/hultenvp/solis-sensor/issues/369

changeofaspect, 2024-06-07:
I have the same issue and it started with the update to HA 2024.6.0.

```
Logger: homeassistant.util.loop
Source: util/loop.py:84
First occurred: 13:31:53 (3 occurrences)
Last logged: 13:39:02

Detected blocking call to open inside the event loop by custom integration 'solis' at custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file: (offender: /config/custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:), please create a bug report at https://github.com/hultenvp/solis-sensor/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 447, in _async_setup_component await asyncio.gather( File "/usr/src/homeassistant/homeassistant/setup.py", line 449, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solis/__init__.py", line 82, in async_setup_entry portal_config = SoliscloudConfig( File "/config/custom_components/solis/soliscloud_api.py", line 185, in __init__ with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:
Detected blocking call to open inside the event loop by custom integration 'solcast_solar' at custom_components/solcast_solar/solcastapi.py, line 230: with open(self._filename) as data_file: (offender: /config/custom_components/solcast_solar/solcastapi.py, line 230: with open(self._filename) as data_file:), please create a bug report at https://github.com/BJReplay/ha-solcast-solar/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solcast_solar/__init__.py", line 110, in async_setup_entry await solcast.load_saved_data() File "/config/custom_components/solcast_solar/solcastapi.py", line 230, in load_saved_data with open(self._filename) as data_file:
Detected blocking call to open inside the event loop by custom integration 'solcast_solar' at custom_components/solcast_solar/solcastapi.py, line 97: with open(self._filename, "w") as f: (offender: /config/custom_components/solcast_solar/solcastapi.py, line 97: with open(self._filename, "w") as f:), please create a bug report at https://github.com/BJReplay/ha-solcast-solar/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever [... cut ...] self.solcast.http_data(dopast=False) File "/config/custom_components/solcast_solar/solcastapi.py", line 543, in http_data await self.serialize_data() File "/config/custom_components/solcast_solar/solcastapi.py", line 97, in serialize_data with open(self._filename, "w") as f:
```

My Solcast integration seems to be affected at the same time with a similar warning and they are referencing a link to handling time zones:  https://developers.home-assistant.io/blog/2024/05/19/fix_zoneinfo_blocking_io/

pivlersmom, 2024-06-10:
Same issue here. Integration is not loading anymore

dpgh947, 2024-06-14:
I am getting this since updating to 2024.6.2 - I also had the same for the alexa_media integration, but that has already been fixed. I get the error at HA restart, but the solis integration does seem to load and work OK after that.

changeofaspect, 2024-06-14:
I had similar with the Solcast integration which generated a message at start up and a slightly different message every time I polled the Solcast server. The warning message for the Solis integration points to Line 185 of the soliscloudapi.py file which is 
**with open('/config/custom_components/solis/workarounds.yaml', 'r')** and seems to be trying to open a file in the event loop. 

The Solcast integration has now been fixed by using   **'async with aiofiles.open'**   instead of   **'with open'**   but it was not quite as simple as that. I am not a Python coder but I noted that other changes were required to be made to make it all work. It will need a Python expert to work through it and make the changes - I am definitely not a Python expert. Also, the reference above to the zoneinfo blocking turned out to be a bit of a red herring, yes, it needed sorting, but it was not the cause of the warning messages which were caused by the **'with open'**.

cfarm31, 2024-06-16:
Hi all 

I'm getting the same error and notes on work grounds or anything like that? Below is my error 

Logger: homeassistant.util.loop
Source: util/loop.py:84
First occurred: 23:06:31 (3 occurrences)
Last logged: 23:07:06

Detected blocking call to open inside the event loop by custom integration 'solis' at custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file: (offender: /config/custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:), please create a bug report at https://github.com/hultenvp/solis-sensor/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 447, in _async_setup_component await asyncio.gather( File "/usr/src/homeassistant/homeassistant/setup.py", line 449, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solis/__init__.py", line 82, in async_setup_entry portal_config = SoliscloudConfig( File "/config/custom_components/solis/soliscloud_api.py", line 185, in __init__ with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:
Detected blocking call to open inside the event loop by custom integration 'solcast_solar' at custom_components/solcast_solar/solcastapi.py, line 230: with open(self._filename) as data_file: (offender: /config/custom_components/solcast_solar/solcastapi.py, line 230: with open(self._filename) as data_file:), please create a bug report at https://github.com/oziee/ha-solcast-solar/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/solcast_solar/__init__.py", line 109, in async_setup_entry await solcast.load_saved_data() File "/config/custom_components/solcast_solar/solcastapi.py", line 230, in load_saved_data with open(self._filename) as data_file:
Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender: /config/custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json"))), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() Fi [... cut ...] c_refresh_modbus_data update_result = await self.async_read_modbus_data(group) File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data res = await self.async_read_modbus_registers_all(group) File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all self.loadLocalData() File "/config/custom_components/solax_modbus/__init__.py", line 332, in loadLocalData fp = open(self._hass.config.path(f"{self.name}_data.json"))

Dino0005, 2024-07-09:
> I had similar with the Solcast integration which generated a message at start up and a slightly different message every time I polled the Solcast server. The warning message for the Solis [...]

to add aiofiles do you have to modify manifest.json?

```
{
  "domain": "solis",
  "version": "3.5.2",
  "config_flow": true,
  "name": "Solis Inverter",
  "iot_class": "cloud_push",
  "documentation": "https://github.com/hultenvp/solis-sensor/",
  "issue_tracker": "https://github.com/hultenvp/solis-sensor/issues",
  "dependencies": [
    "aiofiles"  // Aggiunta la libreria aiofiles
  ],
  "codeowners": ["@hultenvp"]
}
```

Is it correct to do it this way?

Dino0005, 2024-07-09:
> > I had similar with the Solcast integration which generated a message at start up and a slightly different message every time I polled the Solcast server. The warning message for the Solis integration points to Line 185 of the soliscloudapi.py file which is **with open('/config/custom_components/solis/workarounds.yaml', 'r')** and seems to be trying to open a file in the event loop.
> > The Solcast integration has now been fixed by using **'async with aiofiles.open'** instead of **'with open'** but it was not quite as simple as that. I am not a Python coder but I noted that other changes were required to be made to make it all work. It will need a Python expert to work through it and make the changes - I am definitely not a Python expert. Also, the reference above to the zoneinfo blocking turned out to be a bit of a red herring, yes, it needed sorting, but it was not the cause of the warning messages which were caused by the **'with open'**.
> 
> to add aiofiles do you have to modify manifest.json?
> 
> ```
> {
>   "domain": "solis",
>   "version": "3.5.2",
>   "config_flow": true,
>   "name": "Solis Inverter",
>   "iot_class": "cloud_push",
>   "documentation": "https://github.com/hultenvp/solis-sensor/",
>   "issue_tracker": "https://github.com/hultenvp/solis-sensor/issues",
>   "dependencies": [],
>   "codeowners": ["@hultenvp"],
>   "requirements": [
>     "aiofiles==23.1.0"  // Added aiofiles library
>   ]
> }
> ```
> 
> Is it correct to do it this way?

OK, it seems so

```
{
  "domain": "solis",
  "version": "3.5.2",
  "config_flow": true,
  "name": "Solis Inverter",
  "iot_class": "cloud_polling",
  "documentation": "https://github.com/hultenvp/solis-sensor/",
  "issue_tracker": "https://github.com/hultenvp/solis-sensor/issues",
  "dependencies": [],
  "codeowners": ["@hultenvp"],
  "requirements": [
    "aiofiles>=23.1.0,<24.0.0"
  ]
}

```
after editing soliscloud_api.py, after restarting Home Asssitant, I no longer have the warning for "Detected blocking call to open inside the event loop by custom integration solis"

jmason, 2024-07-09:
I don't think there's any need to add aiofiles as a requirement; as far as I can tell it seems to be already loaded.  However I think there were some other tricky issues, and https://github.com/hultenvp/solis-sensor/pull/380 is a pull request to fix the issue which seems to work in my install at least.

jmason, 2024-07-10:
@Dino0005 after restarting HA with your changes, was the solis integration actually loaded? during my testing, with the async code invocation inside soliscloud_api.py, HA restarted ok and the error message about "Detected blocking call to open inside the event loop by custom integration solis" no longer appeared -- but in fact the solis integration had been disabled due to the syntax error of attempting to use async code in a place that didn't support it.  (I had to lower the logging level to make this error visible in the logs.)

Dino0005, 2024-07-10:
Yes, it's working

[image]

but my first attempt with this manifest.json: 
```
{
  "domain": "solis",
  "version": "3.5.2",
  "config_flow": true,
  "name": "Solis Inverter",
  "iot_class": "cloud_push",
  "documentation": "https://github.com/hultenvp/solis-sensor/",
  "issue_tracker": "https://github.com/hultenvp/solis-sensor/issues",
  "dependencies": [],
  "codeowners": ["@hultenvp"],
  "requirements": [
    "aiofiles==23.1.0"  // Added aiofiles library
  ]
}
```
gave me an error saying the integration was not found and was not loaded. So I reverted manifest.json and soliscloud_api.py from v.3.5.2.

So I tried to replace only manifest.json...rewriting it like this (maybe I had made some typing errors in the first manifest.json)
{
  "domain": "solis",
  "version": "3.5.2",
  "config_flow": true,
  "name": "Solis Inverter",
  "iot_class": "cloud_polling",
  "documentation": "https://github.com/hultenvp/solis-sensor/",
  "issue_tracker": "https://github.com/hultenvp/solis-sensor/issues",
  "dependencies": [],
  "codeowners": ["@hultenvp"],
  "requirements": [
    "aiofiles>=23.1.0,<24.0.0"
  ]
}

and the integration was loaded. After I was sure of this I replaced soliscloud_api.py and when I restarted HA the integration gave no problems.

In addition to manifest.json and soliscloud_api.py, the previous day I had also replaced __init__.py to fix deprecation warnings.

I'll attach the 3 files if you want to take a look.
[Fix 1.zip](https://github.com/user-attachments/files/16162483/Fix.1.zip)

jmason, 2024-07-11:
Thanks! that works perfectly -- much nicer than my version :)

Also, I think you're correct that "iot_polling" is a more appropriate iot_class.
And this fixes https://github.com/hultenvp/solis-sensor/issues/379 too, right?

Dino0005, 2024-07-12:
Yes, __init__.py fixes deprecation warnings #379

closed 2024-08-27 by hultenvp