hultenvp/solis-sensor #378: bug

hannes3510, 2024-06-25:
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:325
integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 16:01:03 (1 occurrences)
Last logged: 16:01:03

Entity sensor.solis_daily_grid_energy_used from integration solis has state class total_increasing, but its state is negative. Triggered by state -7.0 with last_updated set to 2024-06-25T13:54:59.999999+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues

swests, 2024-08-14:
I'm seeing this too.
HA 2024.7.4
Solis_Sensor: v3.5.2

```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:326
integration: Sensor (documentation, issues)
First occurred: 00:12:12 (2 occurrences)
Last logged: 00:15:10

Entity sensor.solis_daily_grid_energy_used from integration solis has state class total_increasing, but its state is negative. Triggered by state 0.0 with last_updated set to 2024-08-13T23:02:12.642384+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.solis_daily_grid_energy_used_2 from integration solis has state class total_increasing, but its state is negative. Triggered by state -0.01 with last_updated set to 2024-08-13T23:12:15.035449+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues

```

hultenvp, 2024-08-28:
This is a bug in the SolisCloud where they sometimes have errors in calculating the energy used.
If I change the state class then your energy dash will mess up, so nothing I can do.
Please report a bug with Solis.

closed 2024-08-28 by hultenvp