hultenvp/solis-sensor #379: async_forward_entry_setup is deprecated and will stop working in Home Assistant 2025.6

Dino0005, 2024-07-04:
**Describe the bug**
A clear and concise description of what the bug is.
async_forward_entry_setup is deprecated and will stop working in Home Assistant 2025.6

**Versions**
Versions
Home Assistant 2024.7.0
HACS Version: None (manual installation of Solis integration)
Integration version: 3.5.2

  logs:

Registratore: homeassistant.helpers.frame
Fonte: helpers/frame.py:151
Prima occorrenza: 3 luglio 2024 alle ore 21:57:41 (1 occorrenze)
Ultima registrazione: 3 luglio 2024 alle ore 21:57:41

Detected code that calls async_forward_entry_setup for integration solis with title: Station 159E83 and entry_id: 00bb89d37c07b68a42b5be5326da28a6, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.

Registratore: homeassistant.helpers.frame
Fonte: helpers/frame.py:184
Prima occorrenza: 3 luglio 2024 alle ore 21:57:26 (1 occorrenze)
Ultima registrazione: 3 luglio 2024 alle ore 21:57:26

Detected that custom integration 'solis' calls async_forward_entry_setup for integration, solis with title: Station 159E83 and entry_id: 00bb89d37c07b68a42b5be5326da28a6, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/solis/__init__.py, line 94: hass.async_create_task(, please create a bug report at https://github.com/hultenvp/solis-sensor/issues

**Screenshots**
[image]
[image]

Dino0005, 2024-07-09:
This is the __init__.py file with fixes for async_forward_entry_setup which is deprecated
If it helps anyone, here it is
[__init__.py.zip](https://github.com/user-attachments/files/16146533/__init__.py.zip)

L81ker, 2024-08-25:
I see similar errors, but not sure if these are the same root cause. My log has these entries relating to Solis:

2024-08-25 10:25:09.139 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/solis/workarounds.yaml', 'r') inside the event loop by custom integration 'solis' at custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file: (offender: /config/custom_components/solis/soliscloud_api.py, line 185: with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:), please create a bug report at https://github.com/hultenvp/solis-sensor/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component
    result = await _async_setup_component(hass, domain, config)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 461, in _async_setup_component
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 463, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 752, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solis/__init__.py", line 82, in async_setup_entry
    portal_config = SoliscloudConfig(
  File "/config/custom_components/solis/soliscloud_api.py", line 185, in __init__
    with open('/config/custom_components/solis/workarounds.yaml', 'r') as file:

2024-08-25 10:25:09.143 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'solis' calls async_forward_entry_setup for integration, solis with title: Station 12EF18 and entry_id: 87db3c31b69eda53ecb87b0569535cee, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/solis/__init__.py, line 94: hass.async_create_task(, please create a bug report at https://github.com/hultenvp/solis-sensor/issues

Dino0005, 2024-08-25:
The issue you are having is this:
https://github.com/hultenvp/solis-sensor/issues/377