hultenvp/solis-sensor #382: Variable sensor.solis_energy_today not updated correctly!

Bildhauer1963, 2024-07-19:
The variable solis energy today (sensor.solis_energy_today) is ist not updated correctly. Starting at 12.00 am it seems that it starts updating with an initial value of the cumulated value for the day. Please look at 
[image]. Solis itself show for today the following dat 
[image]

hultenvp, 2024-08-28:
Most Solis inverters reset at dawn, meaning they continue reporting energy stats from the previous day until dawn via the API.
There is a filter in the code that ignores data based on inverter state during the morning. Looks like your inverter reports the wrong state via the API, meaning proper reporting only starts at noon. 
Please provide a screenshot of inverter state from midnight till somewhere after noon for a day that your observe this anomaly

eddysteurs, 2024-10-07:
I have the same problem, is has nothing to do with inverter reset at dawn, it occurs everytime you restart home assistant, it gets value 'unknown' instead of zero (like the month and year variables do get). I think it is because we have a battery only converter so there are no energy values (never), I also have to manually change it fro munknown to zero after each HA restart

J3G0, 2024-10-30:
[image]

I think I have the same issue, it shows a big peak of the previous day's worth of solar production.

J3G0, 2024-10-30:
Also (sorry I'm new to HA), is there any way to see a more detailed view instead of the 1 hour blocks in the energy dashboard.

J3G0, 2024-11-11:
[image]
[image]

As you can see the previous days worth of energy is added right when the inverter is initializing.

J3G0, 2024-11-15:
It looks like when the inverter is starting up the previous days total is still in the system and get pushed. However it corrects itself shortly after, on the next update it seems. This is also seen in the graph. The solis energy today also recorrects itself. Look like there needs to be a filter implemented to prevent this inverter behaviour to get stuck in the energy dashboard.

Some more screens;
[image]
[image]
[image]

hultenvp, 2024-11-17:
Might have the same root cause as #395 .

Please try again with release 3.7.1

J3G0, 2024-11-24:
Looks like in the changelog you defined the minimum required HA v to be v2024.9.0.

I am using the Synology HA core version, which is currently HA v2023.7.3-22.

So I can't update the integration. Any reason that requires the 2024.9 version?

J3G0, 2024-11-28:
I moved to a docker installation and V2024.9.3.

Yesterday no power really but today some.

Will update if I see that v3.7.1 fixed it :)

[image]

> Might have the same root cause as #395 .
> 
> Please try again with release 3.7.1

J3G0, 2024-11-29:
> Might have the same root cause as #395 .
> 
> Please try again with release 3.7.1

Sadly the 3.7.1 release doesn't fix this problem. This morning I got the full production of yesterday added again.
[image]

J3G0, 2025-01-03:
Still happening in 3.8.1 as well, a check should be implemented to check how long the inverter is online for. To exclude this false value from being included in the reporting.
[image]

J3G0, 2025-01-25:
Update from my side.

I looked at the problem from a mathematical side. Solis cloud gives you an update of the actual power delivered every 5 minutes. The solid power today entity has the glitch but the current power is accurate.

So instead of giving the energy integration the solid power today entity I decided to provide it with an integral that sums the actual power generated to have an overview of the kWh that were delivered. 

This has a couple benefits, the glitch where yesterday's power is counted at the initialisation of the inverter is not carried over. Secondly since we use the actual power, even low amounts of power are also counted. Where normally you don't see any graph updates if there is less than 0.1 kWh generated.

I'm my case I used 3 decimal precision. Now I get kWh numbers with a precision of 0.005 kWh. So far the accuracy seems to hold up.

How?
- Riemann integral with Solis AC output as input. This then provides a kWh metric (left oriented integral, this underestimates a bit but better than over estimating.)
- Secondly you need the utility helper to make it a daily counter.

This utility helper you then add as the solar production in the energy dashboard.

It's not a fix but a containment that has some advantages, at least in precision.

See also the screens below 👇 
Hope this helps people facing the same problem 🤗

[image]
[image]
[image]
[image]

hultenvp, 2025-02-08:
Thanks, how does this work when HA is restarted during the day?
The integration is stateless, so would lose all history for the part of the day before getting back online.
Or does the riemann integration jst continue where it left (but then still you could have a gap if the system was down for a longer time)?

J3G0, 2025-02-08:
> Thanks, how does this work when HA is restarted during the day? The integration is stateless, so would lose all history for the part of the day before getting back online. Or does the riemann [...]

I assume that it stops counting your production as it uses the 5 minuten intervals of the actual power production in W to count the kWh. So it would cause a zero production during the off time. But having restarted often throughout the last week's to do updates I've found it to be more reliable than the actual kWh solis information.

Why? Well the solis integration only gives you 0.1 kWh intervals for the day and only full kWh for the month. As an example I have also made a month counter to count the production per month. So the solis on day basis alway has +-0.1 kW deviation. The monthly production always has a +- 1 kWh deviation.

[image]
[image]

Todays overview;

[image]

The prediction function also is very close to actual production. 

Off course, this is an interpolation running on 5 minute data intervals. But it seems that is accurate enough on 5 minute intervals. The Riemann approximation is always within the deviation of the solis anyway. That at least gives me enough confidence to use this method. And I like that you get a smooth graph compared to the blocky 0.1 kWh updates from solis.

I tried contacting solis to increase the data push to 1 minute, they told me they could not provide this anymore due to server limitations. That would have made it more accurate.

J3G0, 2025-02-08:
Today overview solis;

[image]

Today overview Riemann ;

[image]