hultenvp/solis-sensor #383: sensor.solis_energy_today stopped working (becomes unknown)

khangwei, 2024-08-27:
Starting today (27 August)

sensor.solis_energy_today stopped working. I am not sure why. No value is reported . When I checked my Solis App and tried to tally against the entities on my HA, I realized this value is now reported under sensor.solis_daily_grid_energy_used which sounds wrong to begin with.

I assume nothing in this integration changed, so it must be at Solis API...

Anyone having the same issue as me?

Thank you

jmason, 2024-08-27:
It's not across all users, at least -- I'm still seeing correct values under solis_energy_today.... odd!

hultenvp, 2024-08-28:
Same here, still working. 
Please check the logs and see what the test script does
https://github.com/hultenvp/solis-sensor/tree/master/test

khangwei, 2024-08-28:
thanks for your reply. sorry for the silly question but how do i use the test script?

hultenvp, 2024-08-28:
The doc says look at the [wiki](https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues) ;-)

wiikka, 2024-08-29:
For me everything worked until this morning, and now integration doesn't work anymore. I haven't made any updates nor restarts between last evening (when everything worked) vs this morning (when stopped to work)

CraigCallender, 2024-08-29:
Looks like the issue (for me) is that no inverters are found:
[image]

And it'll be somewhere in here.  I'm taking a look at this function to see what I can find out:
```python
async def fetch_inverter_list(self, plant_id: str) -> dict[str, str]:
"""
Fetch return list of inverters { inverter serial : device_id }
"""

device_ids = {}

params = {
	'stationId': plant_id
}
result = await self._post_data_json('/v1/api/inverterList', params)

if result[SUCCESS] is True:
	result_json: dict = result[CONTENT]
	if result_json['code'] != '0':
		_LOGGER.info("%s responded with error: %s:%s",INVERTER_DETAIL, \
			result_json['code'], result_json['msg'])
		return device_ids
	try:
		for record in result_json['data']['page']['records']:
			serial = record.get('sn')
			device_id = record.get('id')
			device_ids[serial] = device_id
	except TypeError:
		_LOGGER.debug("Response contains unexpected data: %s", result_json)
elif result[STATUS_CODE] == 408:
	now = datetime.now().strftime("%d-%m-%Y %H:%M GMT")
	_LOGGER.warning("Your system time must be set correctly for this integration \
	to work, your time is %s", now)
return device_ids
```

CraigCallender, 2024-08-29:
Most likely it's a solis_cloud API issue.  Running the test script results in a response from the API on line 67 of (`station_id` and `inverter_sn` have been masked):
```
inverter_detail(station_id= 12345678901234, inverter_sn= 43210987654321):  Failed with: API returned an error: Communication error. Please refresh and try again later, error code: None, response None
```

CraigCallender, 2024-08-29:
I've logged a ticket with Solis Support.  It's often a struggle to get them to forward it to their cloud team (the support email is more for support setting up Inverters, etc...).  But I'll keep the thread updated.  [Ticket ](https://solis-service.solisinverters.com/en/support/tickets/425868) (not sure you'll be able to see it)

hultenvp, 2024-08-29:
Thanks @CraigCallender for following up with them. 
What is odd is that it only fails for a subset of people.
I have no problems at all, for example.
It could be anything on their backend, but I suspect one of their server instances is down.

I closed the other ticket (#384 ), but I'll leave this one open for anyone to update on status. But I consider this an API issue and have labeled to not a bug

ricardito62, 2024-08-29:
Was working for me until this morning (so, while failing for a subset of people, it may be a growing subset…)

deligatedgeek, 2024-08-29:
Same happened to me this morning.  Have tweeted them, but not sure they care that much

conallob, 2024-08-29:
Another user who has no data in Home Assistant as of today, 2024-08-29. SolisCloud v4.3.9does indeed work though.

One interesting observation after I installed V3.6.0 of this integration is I got promoted to re-auth to SolisCloud. I'm unable to do so, as my PlantID appears to be in hexadecimal, but the integration form is looking for a numberic ID, in base10

strickersuk, 2024-08-29:
Definitely a growing subset, me too.

hultenvp, 2024-08-29:
> Another user who has no data in Home Assistant as of today, 2024-08-29. SolisCloud v4.3.9 does indeed work though.
>
> One interesting observation after I installed V3.6.0 of this integration is I got [...]

Hi Conall, please follow the steps in the installation manual. 
https://github.com/hultenvp/solis-sensor
PlantID is the 19-digit number from the plant detail page on SolisCloud.

Maybe reauth is the way to go for everyone observing issues?

strickersuk, 2024-08-29:
I tried to reauth but got an error, for clarification this was using the 19 digit number.

strickersuk, 2024-08-29:
No, tried to reinstall using the existing? Is it worth a service request for a new one?

hultenvp, 2024-08-29:
Not sure if it said before but the apiManage page now explicitly states only for Organizational leaders:
[image]

I sincerely hope they are not in the process of revoking access for end-users

hultenvp, 2024-08-29:
@strickersuk no service request, I only mean the steps on the apiManage page; pressing the activation button, verification mail etc.

hultenvp, 2024-08-29:
This part: 
[image]

strickersuk, 2024-08-29:
Yes, tried that and used my existing API keys.

hultenvp, 2024-08-29:
> Yes, tried that and used my existing API keys.

You did not get a new key and secret?

strickersuk, 2024-08-29:
> > Yes, tried that and used my existing API keys.
> 
> You did not get a new key and secret?

Tried to reauth using the API key and secret and got the same error as OctoStu above.

strickersuk, 2024-08-29:
It just came back.

wiikka, 2024-08-29:
Yes, data again visible in HA!

hultenvp, 2024-08-29:
Somebody at ginlong turned a server off and on again ;-)

dpgh947, 2024-08-29:
Well that was an hour of messing around collecting debug logs that I needn't have bothered with! :-)

I updated to the new 3.6, and it wouldn't connect. Checked the solis app, that was fine. So I collected some logs, and then fell back to previous version, and that didn't start either. Collected some more logs from that, thinking "what has the new version done to my account?" , then realised looking at history that it hadn't worked at all this morning. Then found this thread just as it started working again. Anyway, installed 3.6 again and all seems good now. Great timing!

khangwei, 2024-08-30:
> The doc says look at the [wiki](https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues) ;-)

thank you for this... tried to follow but am getting this error:

[image]

hultenvp, 2024-08-30:
Use [pip](https://python.land/virtual-environments/installing-packages-with-pip) to install the requests package.

khangwei, 2024-08-30:
> Use [pip](https://python.land/virtual-environments/installing-packages-with-pip) to install the requests package.

thank you so much. just tried the test script and all looks OK. 

in verbose mode..

I could find my current generation: "etoday": 18.8, "etoday1": 18.8,

which can be found here : 
[image]

and oddly, the previous entity i used is unknown!
[image]

khangwei, 2024-09-03:
just want to report that everything is now back to normal. I have my solis energy today back... i did not change anything.

crryp, 2024-09-05:
For me this just happend after i did update the solis intergration to the latest version. 

Daily usage = unknown

I read in the comments for some it gets back somehow? Will see if it does for me

The grid usage does show the daily value also mentioned in another comment.

crryp, 2024-09-06:
It went back the next day! 0:00 it was reset to zero!

closed 2024-09-06 by hultenvp