hultenvp/solis-sensor #394: Battery SOC And SOH Drop To Zero Percent Perhaps Once A Day

IM35461, 2024-10-26:
This has been happening now for a few weeks and assumed it was a general fault which would get fixed.

However, you will see in the screen dumps the problem and the issue does not seem to register in the Solis app so not sure of the source of the problem.

Any ideas?

Many thanks
[image]
[image]
[image]
[image]

IM35461, 2024-10-30:
Still seems to be happening :(

[image]

IM35461, 2024-11-04:
I have been looking at the historical data for battery SOC and the first drop to zero happened on the 26th October so something I guess changed about then.

Nobody else having this minor issue?

[image]
[image]
[image]

jmason, 2024-11-08:
nothing here, at least:
[image]

petervanhulten-tomtom, 2024-11-18:
The Soliscloud API has several issues and this is one of them.
This is not an issue of the integration.
Please report to Ginlong Solis.

hultenvp, 2024-11-18:
The Soliscloud API has several issues and this is one of them.
This is not an issue of the integration.
Please report to Ginlong Solis.

IM35461, 2024-11-18:
> The Soliscloud API has several issues and this is one of them. This is not an issue of the integration. Please report to Ginlong Solis.

Many thanks for the info and I was going to post anyway as since the beginning of this month it has not happened so I guess Ginlong Solis must have changed something.

[image]

IM35461, 2024-11-18:
All sorted, request closed.

closed 2025-02-08 by hultenvp