hultenvp/solis-sensor #395: sensor.solis_energy_today reports same (incorrect) value for all three inverters

henrikeri, 2024-11-12:
**Describe the bug**
sensor.solis_energy_today, sensor.solis_energy_today_2 and sensor.solis_energy_today_3 reports the exact same value. It seems they all three might be added together

**Config**
Configured through the GUI with SolisCloud. Keys etc. nothing special selected.

**Versions**
Latest of all (non-beta)

**Debug Log*
Can't see anything meaningful

**Screenshots of inverter soliscloud vs reported in HA for same day (y-day)**
**Inverter 1:**
[image]
[image]

**Inverter 2:**
[image]
[image]

**Inverter 3:**
[image]
[image]

hultenvp, 2024-11-16:
Can you check the energy_today value equals the energy of the whole plant/station?
https://soliscloud.com/#/station

There is a switch to use station instead of inverter for energy_today. Default it is off (i.e. inverter is used)
https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/workarounds.yaml

There might be a bug in the workaround implementation, but let's start checking if it's set to false and if the value equals the plant value.

henrikeri, 2024-11-16:
Hi,

It does indeed equal the energy of the whole station. I also double checked my "workarounds.yaml" and it is set to "false".

Cheers,
H

hultenvp, 2024-11-17:
Thanks for confirming.

I'll have a shot at trying to reproduce.

hultenvp, 2024-11-17:
@henrikeri : I think this should fix it, but I only have one inverter, so can't be 100% sure.
Please reopen if still an issue.

closed 2024-11-17 by hultenvp