hultenvp/solis-sensor #397: Integration not loading after updated on 17th Nov 2024

brian-kehoe, 2024-11-18:
**Describe the bug**
This had been working up until yesterday when I updated.

An error message is now displayed on the Repairs dashboard:
Error Message
The integration solis could not be found. This happens when a (custom) integration was removed from Home Assistant, but there are still configurations for this integration. Please use the buttons below to either remove the previous configurations for solis or ignore this.

**Config**
Cannot access this in the integrations page

**Versions**
HA Version an deployment: 
Core 2024.11.1
Supervisor 2024.11.2
Operating System 13.2
Frontend 20241106.2

HACS version
Integration version: | 2.0.1
Frontend version: | 20240903140523

Integration version
Unknown

**Debug logs**
Adding the debug information did not result in any additional logs being generated. This error and warning is all that is being displayed in the logs relating to 'solis'

Setup failed for 'solis': Integration not found  
Failed to load integration for translation: Integration 'solis' not found

Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**

[image]

[image]

[image]

[image]

[image]

hultenvp, 2024-11-18:
Use the redownload option in HACS and see if that fixes it.
You can also manually download the translation.py from here and add it to the right folder under custom_components

brian-kehoe, 2024-11-18:
Thank you, that fixed it for me.

closed 2024-11-18 by brian-kehoe