hultenvp/solis-sensor #401: Setup failed for custom integration 'solis': Requirements for solis not found: ['aiofiles>=23.1.0,<24.0.0'].

jojkopeter, 2024-12-05:
**Describe the bug**
A clear and concise description of what the bug is.

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment
HACS version
Integration version

Error after update Home Assistant 

Setup failed for custom integration 'solis': Requirements for solis not found: ['aiofiles>=23.1.0,<24.0.0'].

MiguelCosta, 2024-12-05:
Duplicate: https://github.com/hultenvp/solis-sensor/issues/399

IM35461, 2024-12-05:
I am getting the same after updating my system.

Logger: homeassistant.setup
Source: setup.py:269
First occurred: 08:40:19 (1 occurrences)
Last logged: 08:40:19

Setup failed for custom integration 'solis': Requirements for solis not found: ['aiofiles>=23.1.0,<24.0.0'].

terppatyyppi, 2024-12-05:
Yes, same here. Broke after 2024.12 update. Looking forward to some wonderful people creating a fix!

trevore40, 2024-12-05:
I have the same issue too after the latest update.
Please fix it as soon as you can.
Thank you!

los-cuentos, 2024-12-05:
Same issue

ColinSainsbury, 2024-12-05:
If you open \custom_components\solis\manifest.json with Studio Code Server or the Text File Editor and change line 12 to read     "aiofiles>=23.1.0,<25.0.0" (in the broken version it reads     "aiofiles>=23.1.0,<24.0.0") and reboot HA it will work.

The latest version of HA upgraded to the latest version of Python which includes version 24.1.0 of aiofiles which was outside of the range specified thereby breaking the integration.

trevore40, 2024-12-05:
Thanks Colin, I will try it later

IM35461, 2024-12-05:
> If you open \custom_components\solis\manifest.json with Studio Code Server or the Text File Editor and change line 12 to read `"aiofiles>=23.1.0,<25.0.0" ` (in the broken version it reads `"aiofiles>=23.1.0,<24.0.0"`) Reboot HA it will work.
> 
> The latest version of HA upgraded to the latest version of Python which includes version 24.1.0 of aiofiles which was outside of the range specified thereby breaking the integration.

Colin,
Many thanks that did the trick :)

[image]

hultenvp, 2024-12-06:
Fixed, by pr #400 
Release 3.7.2

closed 2024-12-06 by hultenvp