hultenvp/solis-sensor #402: Error with HA Core 2024.12.0

danielTinkiel, 2024-12-05:
**Describe the bug**
When updating HA Core ro version 2024.12.0, HA reports Invalid Configuration for the SolisCloud sensor integration V3.7.1 which is the one I have installed and believed to be updated. in these conditions, all the entities report "unavailable", As soon as I restored to HA Core 2024.11.3 everything returns to work normally.

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment
HACS version
Integration version

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

steveworkman, 2024-12-05:
I'm seeing the same issue - the integration doesn't start in 2024.12.0

ojectan, 2024-12-05:
I had the same problem, I restored the system to the previous version and everything was normal, waiting for feedback to be able to update the Home Assistant.

cskiwi, 2024-12-05:
I have the following error in my logs:

```logs
Logger: homeassistant.setup
Source: setup.py:269
First occurred: 11:04:16 (1 occurrences)
Last logged: 11:04:16

Setup failed for custom integration 'solis': Requirements for solis not found: ['aiofiles>=23.1.0,<24.0.0'].
```

benv666, 2024-12-05:
Duplicate: #399

Zhalias, 2024-12-05:
I had the same problem.

AndersNilsson19, 2024-12-06:
I will wait some days to se if there is a update here, if not i will backup to :)

Elro071, 2024-12-06:
I got the same problem after update reinstall solis cloud but still the same problem

Bonneville68, 2024-12-06:
One more; died directly after update to core 2024.12

bbartje, 2024-12-06:
+1 here too, hopefully a fix soon.
However current production is low anyway so no panic here.

hultenvp, 2024-12-06:
Fixed, by pr #400

closed 2024-12-06 by hultenvp

danielTinkiel, 2024-12-06:
Thank you very much! Now it works fine.

Daniel Tinkiel
10 King Edward Park,
Bray, Co. Wicklow, A98EC52
Ireland
+353 86 382 3929

Closed #402<https://github.com/hultenvp/solis-sensor/issues/402> as completed.