hultenvp/solis-sensor #406: Solis inverter Grid Phase 0 value

DenisTimofijuk, 2024-12-12:
<p><strong>Describe the bug</strong><br>
The Solis integration in Home Assistant shows <code inline="">0.0 W</code> for Solis inverter Grid Phase Power (1, 2, 3 phases), despite accurate values being visible on the official Solis monitoring platform. This indicates the integration is not correctly retrieving or displaying the Grid Phase Power data.</p>
<p><strong>Config</strong></p>
<ul>
<li>Cloud-based Home Assistant integration for Solis monitoring</li>
</ul>
<p><strong>Versions</strong></p>
<ul>
<li>Home Assistant Core: 2024.11.1</li>
<li>Frontend: 20241106.2</li>
<li>Solis integration: 3.7.1</li>
</ul>
<p><strong>Observed Behavior in Sensors</strong></p>

Sensor | Value
-- | --
Solis inverter Grid Phase1 Power | 0.0 W
Solis inverter Grid Phase2 Power | 0.0 W
Solis inverter Grid Phase3 Power | 0.0 W

<p><strong>Expected Behavior</strong><br>
The integration should reflect accurate Grid Phase Power values for the 1, 2, and 3 phases, as visible on the official Solis monitoring platform.</p>

<p><strong>Screenshots</strong><br>
[image]
</p>

hultenvp, 2024-12-27:
Could you run the test app and post the results here (remove your personal details)?
https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues

The API often behaves differently from the web application itself. I want to confirm the API does indeed expose the right data.

DenisTimofijuk, 2025-01-21:
Thank you for your response and for providing the link to the test app. I understand the API can sometimes behave differently from the web application. However, I was hoping that you could help verify this issue on your end.

As the developer or maintainer, you likely already have a test environment set up for such scenarios, whereas I would need to set up everything from scratch. Since this issue affects multiple users, confirming it within your test environment might streamline the process and ensure accurate diagnostics.

At the same time, while testing this on your side, I hope you might spot any new features Solis has added and consider updating your integration to support them as well.

I truly appreciate the effort you’ve put into maintaining this integration and hope this approach could save time for everyone involved. Please let me know if you need any further information or assistance from my side.

omenukas, 2025-01-28:
Is there any hope that this problem will be fixed? I always install all updates and I haven't seen this data since April last year. Since I see that this problem is not unique to me, I am still waiting for this problem to be fixed. However, as far as I can see, nothing changes. Today I tried to roll back to 3.5.1 and it started showing all the parameters as before.

omenukas, 2025-01-31:
Sorry, but I'm completely new to Python. I can do something, just follow the instructions

DenisTimofijuk, 2025-01-31:
Same here. I only know JavaScript. I am good at Python as much as ChatGPT will be able to write code for me. If you could give some instructions or steps on what to use and what to achieve, I could write some tests using ChatGPT help. I tried to read the Solis API documentation but one I had was quite old and the code examples are written in Java and not quite clear to me...

DenisTimofijuk, 2025-01-31:
Running these tests I in VSCode I am getting fallowing errors and I have no idea how to fix them:
```
[Running] python -u "e:\Projects\SOLIS\solis-sensor\test\apitest_async.py"
Traceback (most recent call last):
  File "e:\Projects\SOLIS\solis-sensor\test\apitest_async.py", line 8, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

[Done] exited with code=1 in 1.188 seconds

[Running] python -u "e:\Projects\SOLIS\solis-sensor\test\tempCodeRunnerFile.py"
Traceback (most recent call last):
  File "e:\Projects\SOLIS\solis-sensor\test\tempCodeRunnerFile.py", line 1, in <module>
    aiohttp
NameError: name 'aiohttp' is not defined

[Done] exited with code=1 in 0.544 seconds
```

DenisTimofijuk, 2025-01-31:
Results running CanonicalizedResource = "/v1/api/inverterDetailList"
This json result has 814 lines:

```
{
    "code": "0",
    "data": {
        "current": 1,
        "pages": 1,
        "records": [
            {
                "aLookedPower": 0,
                "aPhasePowerFactor": 0,
                "aReactivePower": 0,
                "acCoupledMonthEnergy": 0.0,
                "acCoupledMonthEnergyStr": "kWh",
                "acCoupledTodayEnergy": 0.0,
                "acCoupledTodayEnergyStr": "kWh",
                "acCoupledTotalEnergy": 0.0,
                "acCoupledTotalEnergyStr": "kWh",
                "acCoupledYearEnergy": 0.0,
                "acCoupledYearEnergyStr": "kWh",
                "acGridLookedPowerA": 110.0,
                "acGridLookedPowerB": 440.0,
                "acGridLookedPowerC": 50.0,
                "acGridPowerA": 100.0,
                "acGridPowerB": 440.0,
                "acGridPowerC": -20.0,
                "acGridReactivePowerA": 50.0,
                "acGridReactivePowerB": 30.0,
                "acGridReactivePowerC": -60.0,
                "acInType": 0,
                "acOutputType": 3,
                "afciDataFlag": 0,
                "afciType": "0",
                "afciTypeStr": "AFCI0",
                "afciVer": "0",
                "afciVersionAll": "0000",
                "alarmState": 1,
                "allInCome": 521.664,
                "ammeterId": "*********************",
                "apparentPower": 680.0,
                "apparentPowerStr": "VA",
                "averagePowerFactor": 0,
                "bLookedPower": 0,
                "bPhasePowerFactor": 0,
                "bReactivePower": 0,
                "backup2LookedPower": 0.0,
                "backup2LookedPowerA": 0.0,
                "backup2LookedPowerB": 0.0,
                "backup2LookedPowerC": 0.0,
                "backup2LookedPowerOriginal": 0.0,
                "backup2LookedPowerStr": "kVA",
                "backup2MonthEnergy": 0.0,
                "backup2MonthEnergyStr": "kWh",
                "backup2Power": 0.0,
                "backup2PowerA": 0.0,
                "backup2PowerB": 0.0,
                "backup2PowerC": 0.0,
                "backup2PowerOriginal": 0.0,
                "backup2PowerStr": "kW",
                "backup2TodayEnergy": 0.0,
                "backup2TodayEnergyStr": "kWh",
                "backup2TotalEnergy": 0.0,
                "backup2TotalEnergyStr": "kWh",
                "backup2YearEnergy": 0.0,
                "backup2YearEnergyStr": "kWh",
                "backupLookedPower": 0.2,
                "backupLookedPowerA": 100.0,
                "backupLookedPowerB": 30.0,
[... 724 lines cut ...]
                "uPv9": 0,
                "uPv9Str": "V",
                "updateShelfEndTime": 2036937600000,
                "updateShelfEndTimeStr": "2034-07-20",
                "upvTotal": 0,
                "upvTotalStr": "V",
                "userId": "*********************",
                "variable1": 0.0,
                "variable11": 0.0,
                "variable12": 0.0,
                "variable13": 0.0,
                "variable2": 0.0,
                "version": "120006",
                "version2": "080007",
                "warningInfoData": 512,
                "yearInCome": 6.27
            }
        ],
        "size": 20,
        "total": 1
    },
    "msg": "success",
    "success": true
}
```

DenisTimofijuk, 2025-02-01:
And here are results regarding my issue about active power.
Running `CanonicalizedResource = "/v1/api/inverterDetailList"` I am getting these values:
```
"gridPower": -0.865,
"gridPowerA": -205.0,
"gridPowerB": -574.0,
"gridPowerC": -83.0,
```
Image:

[image]

But running `# CanonicalizedResource = "/v1/api/inveterDetail"` I do not get them.

So my advice would be to check things out, see what has changed on the Solis side, see what has been added and update this integration accordingly.

hultenvp, 2025-02-01:
Thanks @DenisTimofijuk for posting this.

inveterDetail (without the 'r') is an old endpoint from API spec version 1 times, they corrected that in spec version 2.0, but left the old endpoint available for backward compatibility. Both work and both give the same response.

Some initial responses, need to dive into it much further:
* acGridPower [fields are not specified](https://github.com/hultenvp/solis-sensor/blob/master/doc/SolisCloud%20Platform%20API%20Document%20V2.0.pdf) and are also not showing up in responses from my own inverter (single phase, no battery)
* also in your responses, similar to mine, "pA" is always zero (this is the one used in the integration), so I still cannot correlate that to @omenukas  statement that if he reverts to 3.5.1 all is working again. Because that would mean pA (and pB, pC) should be working in inverterDetail. However, as it looks you ran the test after sunset? I see no voltage on your strings, so that might also explain why pA is zero in your case. Can you confirm?
* On my system I do not have any fields that show me phase 1 power. The only thing pAC, but that's over all 3 phases if you have 3. 

So it appears the spec and reality are no longer in line with each other and there seems no uniform source for phase-specific power in the API. 
I checked the spec, but v2.0 is still the latest version.

omenukas, 2025-02-01:
@hultenvp I did an experiment especially for you today and changed the versions. I currently had version 3.5.1 and it shows all settings:

[image]
[image]

Then I changed to the latest version 3.8.1 and the grid data no longer shows:

[image]
[image]

I finally rolled back to 3.5.1 and it started showing up again:

[image]

I did all the version changes in the same system with the same inverter and within half an hour.

DenisTimofijuk, 2025-02-01:
@hultenvp I would expect the API to return different results depending on the hardware. So having hardcoded entities is not ideal (although I don't know how HA handles entities on it's side). I would create dynamic entities based on API returns, if that is compatible with HA. This would allow all users to have what they can actually have depending on their hardware...

DenisTimofijuk, 2025-02-01:
@hultenvp if it helps I did testing one more time.

`CanonicalizedResource = "/v1/api/inverterDetailList"`

[image]

From a quick glance I can see that it has everything I need and would like to use in HA.

```
{
    "code": "0",
    "data": {
        "current": 1,
        "pages": 1,
        "records": [
            {
                "aLookedPower": 0,
                "aPhasePowerFactor": 0,
                "aReactivePower": 0,
                "acCoupledMonthEnergy": 0.0,
                "acCoupledMonthEnergyStr": "kWh",
                "acCoupledTodayEnergy": 0.0,
                "acCoupledTodayEnergyStr": "kWh",
                "acCoupledTotalEnergy": 0.0,
                "acCoupledTotalEnergyStr": "kWh",
                "acCoupledYearEnergy": 0.0,
                "acCoupledYearEnergyStr": "kWh",
                "acGridLookedPowerA": 120.0,
                "acGridLookedPowerB": 180.0,
                "acGridLookedPowerC": 50.0,
                "acGridPowerA": 100.0,
                "acGridPowerB": 180.0,
                "acGridPowerC": 20.0,
                "acGridReactivePowerA": 50.0,
                "acGridReactivePowerB": 30.0,
                "acGridReactivePowerC": -50.0,
                "acInType": 0,
                "acOutputType": 3,
                "afciDataFlag": 0,
                "afciType": "0",
                "afciTypeStr": "AFCI0",
                "afciVer": "0",
                "afciVersionAll": "0000",
                "alarmState": 1,
                "allInCome": 521.928,
                "ammeterId": "*****",
                "apparentPower": 480.0,
                "apparentPowerStr": "VA",
                "averagePowerFactor": 0,
                "bLookedPower": 0,
                "bPhasePowerFactor": 0,
                "bReactivePower": 0,
                "backup2LookedPower": 0.0,
                "backup2LookedPowerA": 0.0,
                "backup2LookedPowerB": 0.0,
                "backup2LookedPowerC": 0.0,
                "backup2LookedPowerOriginal": 0.0,
                "backup2LookedPowerStr": "kVA",
                "backup2MonthEnergy": 0.0,
                "backup2MonthEnergyStr": "kWh",
                "backup2Power": 0.0,
                "backup2PowerA": 0.0,
                "backup2PowerB": 0.0,
                "backup2PowerC": 0.0,
                "backup2PowerOriginal": 0.0,
                "backup2PowerStr": "kW",
                "backup2TodayEnergy": 0.0,
                "backup2TodayEnergyStr": "kWh",
                "backup2TotalEnergy": 0.0,
                "backup2TotalEnergyStr": "kWh",
[... 729 lines cut ...]
                "uPv9": 0,
                "uPv9Str": "V",
                "updateShelfEndTime": 2036937600000,
                "updateShelfEndTimeStr": "2034-07-20",
                "upvTotal": 0,
                "upvTotalStr": "V",
                "userId": "*****",
                "variable1": 0.0,
                "variable11": 0.0,
                "variable12": 0.0,
                "variable13": 0.0,
                "variable2": 0.0,
                "version": "120006",
                "version2": "080007",
                "warningInfoData": 512,
                "yearInCome": 6.534
            }
        ],
        "size": 20,
        "total": 1
    },
    "msg": "success",
    "success": true
}
```

hultenvp, 2025-02-02:
> [@hultenvp](https://github.com/hultenvp) I would expect the API to return different results depending on the hardware. So having hardcoded entities is not ideal (although I don't know how HA handles entities on it's side). I would create dynamic entities based on API returns, if that is compatible with HA. This would allow all users to have what they can actually have, depending on their hardware... And you would avoid constantly checking and updating your integration with every API change and trying to make it universal for every inverter...

I would expect not, that's not how the API spec is set up in the sense that the variability is encoded in the spec. For example acOutputType return value specified whether the inverter is single or three-phase. dcInputType tells how many strings/channels  are connected, type field specifies whether the inverter has a battery or not. What I cannot work with is Solis not following their own spec and putting data in different, unspecified fields or them not fully implementing their own spec and causing regression all the time. I'm still not sure whether they have real quality issues or whether it is on purpose to frustrate these kind of initiatives.
We've seen everything already from daily energy not resetting at midnight but in the morning, precision of values changing to whole integers only, wrong units, values only updated in one call and not the other etc.
So yes, if they follow their own spec I can very easily define a set of sensors for each model and expose those to HA, but the reality is that it's a waterbed and a constant puzzle to find the right calls and values that still do offer the needed info. And once we have it right then Solis changes it again or has regression and the hunt starts all over.

I'm working on a full rewrite of the integration, giving end users more control over the actual source of a sensor, and getting rid of a load of legacy from the initial Ginlong API where this whole integration started with. But that's only progressing at a slow pace and will come with its own challenges of being not backward compatible.

lynxus, 2025-02-02:
> I would expect not, that's not how the API spec is set up in the sense that the variability is encoded in the spec. For example acOutputType return value specified whether the inverter is single or [...]

You're not wrong!

I ended up writing my own dashboard to get solar information using php and bootstrap with an sql backend.
It seems that every few months, something changes and i need to rework something from the API.

What i ended up doing to mitigate so many issues was to store everything into a DB and timestamp it at that time.

Then my frontends can just to my DB and know the data will also be correct. (or null/0 if the api has broken)

Annoyingly, writing middleware to ensure the data is always as expected for my frontend. Somewhat of an API for an API 😟

https://solar.grahamstest.com/

DenisTimofijuk, 2025-02-02:
> I would expect not, that's not how the API spec is set up in the sense that the variability is encoded in the spec. For example acOutputType return value specified whether the inverter is single or [...]

Have you tried contacting the Solis team? since they are providing legitimate rights to use their API, you as a developer have all the rights to address these issues and demand that they have a stable API, or at least that they constantly update you with any changes they make, because as we know, any small change on their side can break things on our side. What is the point of using an unstable API?

hultenvp, 2025-02-08:
I cannot validate all the different configurations. It's up to end users to report with Solis and confirm if it got fixed.

For your particular issues I'm going to revert to using InverterDetail as according to your test app results the data is available there.

closed 2025-02-08 by hultenvp