hultenvp/solis-sensor #410: Solis Meter

brida91, 2024-12-21:
**Describe the bug**
solis meter  A B C not show current

**Versions**
HA 2024.12.5
HACS 2.0.1
Integration 3.8.0

[image]

matteoa66, 2024-12-26:
Same for me, I rolled back to HA 2024.11.3

hultenvp, 2024-12-27:
@brida91 @matteoa66 Does it work in 11.3? With 3.8.x?

matteoa66, 2024-12-27:
Is ok with:
Ha 2024.11.3
Solis3.6.0
Ciao, Matteo

hultenvp, 2024-12-28:
Thanks @matteoa66 and when you upgrade to 2024.12.x then the issue occurs?

matteoa66, 2024-12-28:
Yes
Matteo

hultenvp, 2024-12-30:
That is odd.
There should not be any breaking changes in home assistant that only affect those attributes and not the others. 

Still going to investigate. Might be related to new unit enums

hultenvp, 2024-12-31:
Meter item A is only supported in soliscloud API v2.0.
The integration only supports v1.2. For some reason that eludes me the 1.2 API still does report Meter Item A, but always with no value. I'm not sure how you can see values on the meter item A sensor.
Also Meter item A is not specified; it's very unclear what it actually is meant for and how it relates to grid-side AC current.

Do you want to have the AC current of the solis inverter?

Meter Item A Current --> Current of phase on some electricity meter related to the inverter?
AC Current R --> Phase 1 AC current on inverter

hultenvp, 2025-01-06:
@brida91 , @matteoa66 any update? Otherwise I close the issue.

closed 2025-01-09 by hultenvp