hultenvp/solis-sensor #412: Solis integration wont start after update to the last HA version

0x544c, 2025-01-02:
**Describe the bug**
Solis integration wont start after update to the last HA version

**Versions**
Core 2024.12.5
Supervisor 2024.12.0
Operating System 14.1
Frontend 20241127.8

**Debug logs**
/homeassistant/home-assistant.log:
```
2025-01-02 10:34:28.673 ERROR (SyncWorker_5) [homeassistant.util.package] Unable to install package aiofiles>=23.1.0,<24.0.0: × No solution found when resolving dependencies:
  ╰─▶ Because you require aiofiles>=23.1.0,<24.0.0 and aiofiles>=24.1.0, we
      can conclude that your requirements are unsatisfiable.
2025-01-02 10:34:29.333 ERROR (SyncWorker_5) [homeassistant.util.package] Unable to install package aiofiles>=23.1.0,<24.0.0: × No solution found when resolving dependencies:
  ╰─▶ Because you require aiofiles>=23.1.0,<24.0.0 and aiofiles>=24.1.0, we
      can conclude that your requirements are unsatisfiable.
2025-01-02 10:34:29.976 ERROR (SyncWorker_5) [homeassistant.util.package] Unable to install package aiofiles>=23.1.0,<24.0.0: × No solution found when resolving dependencies:
  ╰─▶ Because you require aiofiles>=23.1.0,<24.0.0 and aiofiles>=24.1.0, we
      can conclude that your requirements are unsatisfiable.
2025-01-02 10:34:29.978 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'solis': Requirements for solis not found: ['aiofiles>=23.1.0,<24.0.0'].
```

jmason, 2025-01-05:
Are you on the latest version of this integration? I think this was fixed in https://github.com/hultenvp/solis-sensor/commit/624c1631872926c9fd3b24d637109d6e28290865

hultenvp, 2025-01-06:
Yes, you need to update to 3.8.1.

closed 2025-01-09 by hultenvp