hultenvp/solis-sensor #413: Solis integration not loaded anymore

wtoyoz, 2025-01-04:
Probably after latest update no data come through anymore.
Downloaded older versions did not help.
Reinstalling integration incl API etc did not work either.
Now shows:
no devices or entities
! not loaded

[image]

jmason, 2025-01-05:
Are there any logs from the integration?

Dirkvdsl, 2025-01-05:
Hi,This is the log from HALogger: homeassistant.setupBron: setup.py:334Eerst voorgekomen: 4 januari 2025 om 23:53:55 (1 gebeurtenissen)Laatst gelogd: 4 januari 2025 om 23:53:55Setup failed for custom integration 'solis': Unable to import component: cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)Traceback (most recent call last):  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component    comp = await self.hass.async_add_import_executor_job(           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^        self._get_component, True        ^^^^^^^^^^^^^^^^^^^^^^^^^    )    ^  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run    result = self.fn(*self.args, **self.kwargs)  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component    ComponentProtocol, importlib.import_module(self.pkg_path)                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func    return func(*args, **kwargs)  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module    return _bootstrap._gcd_import(name[level:], package, level)           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed  File "/config/custom_components/solis/__init__.py", line 17, in <module>    from .const import (    ...<8 lines>...    )  File "/config/custom_components/solis/const.py", line 4, in <module>    from homeassistant.components.sensor import (    ...<2 lines>...    )ImportError: cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)During handling of the above exception, another exception occurred:Traceback (most recent call last):  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component    component = await integration.async_get_component()                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component    self._component_future.result()    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component    comp = self._get_component()  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component    ComponentProtocol, importlib.import_module(self.pkg_path)                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func    return func(*args, **kwargs)  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module    return _bootstrap._gcd_import(name[level:], package, level)           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed  File "/config/custom_components/solis/__init__.py", line 17, in <module>    from .const import (    ...<8 lines>...    )  File "/config/custom_components/solis/const.py", line 4, in <module>    from homeassistant.components.sensor import (    ...<2 lines>...    )ImportError: cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)Met vriendelijke groet,Dirk van der Slink

jmason, 2025-01-05:
```
File "/config/custom_components/solis/const.py", line 4, in <module>    from homeassistant.components.sensor import (    ...<2 lines>...    )
ImportError: cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py)
```

That seems odd -- I don't see `STATE_CLASS_MEASUREMENT` mentioned on line 4 of const.py: https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/const.py#L4 -- I wonder if an old version of the integration is still there somehow.

Dirkvdsl, 2025-01-06:
I have more logs from HA

Logger: homeassistant.helpers.typing
Bron: helpers/deprecation.py:222
Eerst voorgekomen: 5 januari 2025 om 23:43:49 (4 gebeurtenissen)
Laatst gelogd: 5 januari 2025 om 23:43:49

HomeAssistantType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please report it to the author of the 'solis' custom integration
ServiceCallType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.ServiceCall instead, please report it to the author of the 'solis' custom integration
HomeAssistantType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.HomeAssistant instead, please create a bug report at https://github.com/hultenvp/solis-sensor/issues
ServiceCallType was used from solis, this is a deprecated alias which will be removed in HA Core 2025.5. Use homeassistant.core.ServiceCall instead, please create a bug report at https://github.com/hultenvp/solis-sensor/issues

⁣Met vriendelijke groet.
Dirk van der Slink
***@***.***​

Dirkvdsl, 2025-01-06:
And another 

Logger: homeassistant.util.loop
Bron: util/loop.py:77
Eerst voorgekomen: 5 januari 2025 om 23:43:49 (1 gebeurtenissen)
Laatst gelogd: 5 januari 2025 om 23:43:49

Detected blocking call to import_module with args ('custom_components.solis',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

⁣Met vriendelijke groet.
Dirk van der Slink
***@***.***​

wtoyoz, 2025-01-06:
It seems everything is working fine again.
Don't know how come.
But i'm glad it's working again.

Running Solis 3.8.1
[image]

jellmers, 2025-01-08:
Still getting the issue.....
[image]
[image]

On the newest versions of HA.. :-(

patduckless, 2025-01-08:
> Still getting the issue..... <img alt="Screenshot 2025-01-08 at 09 00 14" width="1153" src="https://private-user-images.githubusercontent.com/35197861/401078575-32ed71e2-e7aa-4ada-9f97-463b0f21178b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzMjkwMTIsIm5iZiI6MTczNjMyODcxMiwicGF0aCI6Ii8zNTE5Nzg2MS80MDEwNzg1NzUtMzJlZDcxZTItZTdhYS00YWRhLTlmOTctNDYzYjBmMjExNzhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDA5MzE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmZWFhMGUxMmMwMzRlMmUxYWM2OGMwN2Y1ODNiMGY2MjgwOTdhOGFkY2M1MDVlMTZhMjExNWMyZGRlN2VmOTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ZD6lJJ4OGWkE43Lw4KhNhJ5dI79Ml9yulWgeWj-N6v8"> <img alt="Screenshot 2025-01-08 at 08 58 10" width="602" src="https://private-user-images.githubusercontent.com/35197861/401078611-ffb22d61-3be9-4e03-a376-2ff7b7765750.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzMjkwMTIsIm5iZiI6MTczNjMyODcxMiwicGF0aCI6Ii8zNTE5Nzg2MS80MDEwNzg2MTEtZmZiMjJkNjEtM2JlOS00ZTAzLWEzNzYtMmZmN2I3NzY1NzUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDA5MzE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjNGIxNzc2ODg3ZWU4ZmJmNzkxNDAwMjJhNTk3ZWQ5MzQwNDAzYmNjZjQzOThkNWZkZGZhNTc1N2EwMTc5ZWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zUtIV44A1vM1PKViaFYexieZqwP-m0D4SYpMhG-buSk">
> 
> On the newest versions of HA.. :-(

Can you make sure you are on the latest version of the integration please.
[image]

NachtGloei, 2025-01-08:
> Still getting the issue..... <img alt="Screenshot 2025-01-08 at 09 00 14" width="1153" [...]

Had the same issue after updating. After disabling the old version (by putting a _ behind the name of the folder), rebooted HA without Solis, placed the update (again), and then it worked...

SteveShirvo, 2025-01-08:
thank you NachtGloei, your solution worked perfectly

jellmers, 2025-01-09:
Thanks all...  Manually dragged down 3.8.1 and did a restart and all back to normal....

The power of the hive mind is great.

hultenvp, 2025-01-11:
I've seen similar issues coming by in the past years, but never got feedback how they solved it. Thanks for sharing. Seems like a HACS bug where a mismatch between version admin and installed version. 
IIRC there is also a forced reinstall option in HACS. Might also work. 

Closing this issue

closed 2025-01-11 by hultenvp