hultenvp/solis-sensor #414: sensor.solis_energy_this_month.  Home Assistant. State class error

AndersNilsson19, 2025-01-21:
Logger: homeassistant.components.sensor.recorder
Källa: components/sensor/recorder.py:317
integration: Sensor ([dokumentation](https://www.home-assistant.io/integrations/sensor), [ärenden](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Inträffade först: 12:45:10 (1 händelser)
Senast loggade: 12:45:10

Entity sensor.solis_energy_this_month from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 35.0 (36.0) with last_updated set to 2025-01-21T11:41:43.483530+00:00. Please create a bug report at

patduckless, 2025-01-22:
I can confirm the same warning in the logs. 
It's just a warning so not urgent. 
`
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:317
integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 19 January 2025 at 16:25:10 (4 occurrences)
Last logged: 21 January 2025 at 14:20:10

Entity sensor.solis_energy_this_month from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 58.0 (59.0) with last_updated set to 2025-01-19T16:24:37.700750+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.solis_energy_this_year from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 60.0 (62.0) with last_updated set to 2025-01-21T12:02:27.524942+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.solis_energy_total from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 253.0 (255.0) with last_updated set to 2025-01-21T12:02:27.524976+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.solis_daily_grid_energy_used from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 15.72 (16.61) with last_updated set to 2025-01-21T14:16:13.953576+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
`

hultenvp, 2025-01-22:
Not a big in this integration, but a known API issue.  Was gone, apparently they have regression.
SolisCloud API so now and then had issues calculating these kinds of energy figures and came with a new value that is actually less than the previous one. 
So ha complains rightfully, but it should be solved in SolisCloud. 
As Pat says impact is low, it's just a warning something odd is going on.

Will close as not a big, feel free to file an issue with SolisCloud.

closed 2025-01-22 by hultenvp