hultenvp/solis-sensor #422: 'Solis Battery Discharged Today' does not have decimal precision

Skyline-ThomasGH, 2025-02-15:
**Describe the bug**
The 'Solis Battery Discharged Today' entity only increases with full integers, causing very inaccurate energy dashboard graphs. In the past, before I moved my HA instance and updated this integration to the latest version, I did have one digit decimal precision for this entity.

**Config**
(not sure what to put here)

**Versions**
HACS v2.0.5
SolisCloud portal itegration v3.9.0

**Debug logs**
Let me know if I should enable these.

**Screenshots**

[image]

patduckless, 2025-02-20:
Checking the JSON returned from the API it seems that they are sending a float. However my current discharge appears to be a whole number. I will check again tomorrow. 
It may be that Solis send a whole number formated as a float.

patduckless, 2025-02-20:
I have raised this with Solis.

tabannis, 2025-03-12:
This appears to be related to the previous issue #354 and a change made to the API by the Solis cloud gods.

If it helps, the decimal precision data for charge /discharge does still seem to exist in the official app/API...

[image]

hultenvp, 2025-03-14:
This is a fallout of a revert to using inverterDetail instead of inverterDetaillist as there the problem was that the phase metrics were missing.
I prefer to wait for Solis, we could consider  even reverting to the unsupported inveterDetail, but I honestly do not know what we can expect there in terms of regression if we do.

patduckless, 2025-03-28:
> I have raised this with Solis.

Solis never came back to me on this. I chased up last week, with no reply.

mkuthan, 2025-04-05:
> > I have raised this with Solis.
> 
> Solis never came back to me on this. I chased up last week, with no reply.

They didn't answer me either :(

As a workaround I use [integral](https://www.home-assistant.io/integrations/integration/) integration to calculate approximate energy based on power readings.

Create two sensors for power in and out:

```
- name: "Solar Battery Power In"
        unit_of_measurement: W
        device_class: power
        icon: mdi:battery-plus
        state: >
          {% set power = states('sensor.solis_battery_power') | int %}
          {{ power if power > 0 else 0 }}

      - name: Solar Battery Power Out
        unit_of_measurement: W
        device_class: power
        icon: mdi:battery-minus
        state: >
          {% set power = states('sensor.solis_battery_power') | int %}
          {{ -power if power < 0 else 0 }}
```

Two more to estimate the energy in and out:

```
sensor:
  - platform: integration
    source: sensor.solar_battery_power_in
    name: Solar Battery Energy In
    unit_prefix: k
    round: 2
    max_sub_interval:
      minutes: 5
  - platform: integration
    source: sensor.solar_battery_power_out
    name: Solar Battery Energy Out
    unit_prefix: k
    round: 2
    max_sub_interval:
      minutes: 5
```

Voila:

[image]

abandyb, 2025-04-06:
Just a suggestion from a novice for a solution to this problem (I am nowhere near as familiar with the behaviour of the API):

For the Battery Charged and Discharged sensors, could these be populated from https://www.soliscloud.com/api/chart/inverter/day/v2 ?  Or added as secondary sensors?

Inside the request payload are the following:
searchInfo:  "battery_today_discharge_energy,battery_today_charge_energy"
time:  "2025-04-05"
timeZone:  0

and in the response payload, you get an array of values, from which the last entry in each array could be inspected and taken.

Comparing these values with the current integration for yesterday:

[image]

[image]

NB whilst looking this I noticed that there is also an available entry on this API call named "home_load_today_energy" which appears to be the total energy used by the inverter, which I am monitoring to see how accurate it actually is. The Solis App seems to prefer using the 2 rounded battery values as part of its Load calculation which I believe is totally wrong.