hultenvp/solis-sensor #425: Solis plugin does not start/store configuration when at login stale/offline devices are in the device list

marchelh, 2025-02-26:
Solis plugin does not start when at login stale/offline devices are in the device list. The error generated is login failed, but the actual login succeeds. Solution would be to ignore offline devices and and not error out with a login failed, because the login was successful. Also store the config when a login is successful.  

**Config**
It does not come so far, because the config bug out with the login error

**Versions**
Latest provided by HA plugin ( not showing version yet because config page bug out with an error

**Debug logs**
2025-02-26 09:09:13.182 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': False, 'use_energy_today_from_plant': False}
2025-02-26 09:09:13.479 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2025-02-26 09:09:13.479 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['------0000001100', '------0000006000', '------0224140108']
2025-02-26 09:09:13.479 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 4490000000001100
2025-02-26 09:09:14.544 INFO (MainThread) [custom_components.solis.soliscloud_api] /v1/api/inverterDetail responded with error: R0002:The inverter No operation permission or no longer exists
2025-02-26 09:09:15.662 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Unexpected response from server: None
2025-02-26 09:09:15.662 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire plant name, login failed
2025-02-26 09:09:15.662 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire CSRF token

**Screenshots**
If applicable, add screenshots to help explain your problem.

konrad-glowacki, 2025-03-03:
I have the same issue. I cannot add integration. Inverted "649030003000300" is offline and I cannot remove it from solis cloud.

```
2025-03-03 17:29:32 homeassistant  | 2025-03-03 17:29:32.925 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': False, 'use_energy_today_from_plant': False}
2025-03-03 17:29:33 homeassistant  | 2025-03-03 17:29:33.603 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2025-03-03 17:29:33 homeassistant  | 2025-03-03 17:29:33.603 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['649030003000300', '1107A2217080379']
2025-03-03 17:29:33 homeassistant  | 2025-03-03 17:29:33.603 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 649030003000300
2025-03-03 17:29:34 homeassistant  | 2025-03-03 17:29:34.755 INFO (MainThread) [custom_components.solis.soliscloud_api] /v1/api/inverterDetail responded with error: R0002:The inverter No operation permission or no longer exists
2025-03-03 17:29:36 homeassistant  | 2025-03-03 17:29:36.000 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Unexpected response from server: None
2025-03-03 17:29:36 homeassistant  | 2025-03-03 17:29:36.000 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire plant name, login failed
2025-03-03 17:29:36 homeassistant  | 2025-03-03 17:29:36.000 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire CSRF token
```

hultenvp, 2025-03-14:
I have the same myself, but my stale inverter does contain data and uses that to determine it's stale and needs to be ignored.
Yours seems to not report any data at all.

Will have a look, but might need you to test next release to confirm the fix as I cannot reproduce myself.

closed 2025-03-14 by hultenvp

marchelh, 2025-03-14:
Yeh, it looks like I don't have any rights for these devices. Cant even delete them. 

I'm happy to test it before the merge. If you have an branch where you publish your chance them I will check it out.

marchelh, 2025-04-08:
Hi, problem is not solved yet, the new version is now trowing an error:
2025-04-08 07:37:21.367 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': False, 'use_energy_today_from_plant': False}
2025-04-08 07:37:21.587 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2025-04-08 07:37:21.587 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['4490000000001100', '4E80000000006000', '1801060224140108']
2025-04-08 07:37:21.587 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 4490000000001100
2025-04-08 07:37:22.636 INFO (MainThread) [custom_components.solis.soliscloud_api] /v1/api/inverterDetail responded with error: R0002:The inverter No operation permission or no longer exists
2025-04-08 07:37:23.784 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Unexpected response from server: None
2025-04-08 07:37:23.785 INFO (MainThread) [custom_components.solis.soliscloud_api] No access to inverter 4490000000001100, removing
2025-04-08 07:37:23.785 ERROR (MainThread) [aiohttp.server] Error handling request from 127.0.0.1
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 77, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 232, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis/config_flow.py", line 169, in async_step_credentials_secret
    if await api.login(async_get_clientsession(self.hass)):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis/soliscloud_api.py", line 297, in login
    for inv in self._inverter_list.keys():
               ~~~~~~~~~~~~~~~~~~~~~~~~^^
RuntimeError: dictionary changed size during iteration

It looks like the line "No access to inverter 4490000000001100, removing" thats removing the entry from the converter list is generating this exception.