hultenvp/solis-sensor #437: Control API destabilizes remote access to the inverter

mkuthan, 2025-03-20:
I observed that after enabling Control API, the remote access and Monitoring API works unreliably and my inverter went ofline most of the time. My assumption is that with the Control API enabled, the integration is sending too many requests to the Solis Cloud API, and that is the cause of the problems.

Look at data granularity, before and after enabling Control API (at 12:00):

[image]

It looks exactly the same in Solis Cloud:

[image]

I repeated the experiment twice to eliminate any timing issues with the Solis Cloud platform itself. Enabling the Control API in the integration consistently destabilizes remote access to my inverter 😢

jmason, 2025-03-24:
for what it's worth, I'm not seeing the same issue here, and I have the control API enabled --

[image]

Are you seeing anything in the logs?

sjoram, 2025-03-27:
Since enabling the Control as above my inverter keeps going offline. At least every 2 days sometimes every day. I have to unscrew the Wifi module and it will reconnect but this is a pain as it is not that accessible.

Is it possible to disable this function as at the moment the gains of control are not useful if it is offline so often?

jmason, 2025-03-27:
Yes, I think this sounds like a good idea -- or at least do not enable it by default.

We could make it be enabled only if the control password has been set in the config page, does that sound sensible @hultenvp ?

BTW @sjoram is there anything in the logs when the inverter goes offline?

sjoram, 2025-03-27:
Just to add - SolisCloud is just reporting my inverter as being offline. HA is reporting the sensor values that were current when the device went off line. The only reason I know it is offline is that it should be reporting a production kW value but that is still 0 as it went offline when it was dark!

jmason, 2025-03-27:
> HA shows nothing as it is the connection between the inverter and SolisCloud that hangs. It has hung again this morning but I am at work so cannot reboot it or access it. So if there is anything helpful I can do while in this state before I reboot it when I get home let me know!

ah, that's OK -- I was wondering if there were any logs reported by the integration just prior to the disconnection, in case there was some "smoking gun" log of something the integration code might have been doing to trigger it.  Possibly though it's just the increased volume of traffic and/or change in the traffic profile that's triggering it, not a specific API call.

sjoram, 2025-03-27:
I have an vague recollection I changed something somewhere in the integration to try to get it to poll more frequently a few years ago. I can't for the life of me remember where! I wonder if that may be causing more frequent connections than other people and so me more grief. Do you know where the poll interval is stored and what the default is?

mkuthan, 2025-03-27:
I finished very first version of "Solis Cloud Control" integration, focused on controlling inverter only :)

[image]

Functional features:

* ⚡ Storage Modes: "Self-Use", "Feed-In Priority"
* 🛠️ "Battery Reserve" and "Allow Grid Charging" options as Storage Mode attributes
* ⏱️ Charge/Discharge Slots
* 🔋 Battery Reserve SOC, Over Discharge SOC and Force Charge SOC

Non-functional features:

* 📦 **Batch reading of all inverter settings in a single request to fit within the API limits**
* 🔄 Retry logic for API requests to mitigate API stability issues
* 🏡 Best Home Assistant practices for integration development 😜 

I will thoroughly test the extension over the next few days or weeks
https://github.com/mkuthan/solis-cloud-control

jmason, 2025-03-27:
BTW it looks like there are more reports of this issue at https://github.com/fboundy/solis-sensor/issues/8 .

@mkuthan I would be very interested to hear if your new integration avoids this issue!  Are you running that at the same time as the solis-sensor integration is active?

jmason, 2025-03-27:
Aha, I think I may have found the issue; looks like the control code increased frequency of polling to once every minute, instead of once every 5 minutes: https://github.com/hultenvp/solis-sensor/pull/398/files?w=1#diff-af00c6075e98805923b96dcf422d2aab2d26713b9326640d5bf345dfaf6b09ecR38-R39

PR on the way to revert that, let's see if it helps.

mkuthan, 2025-03-28:
> [@mkuthan](https://github.com/mkuthan) I would be very interested to hear if your new integration avoids this issue! Are you running that at the same time as the solis-sensor integration is active?

Yes, both integrations work together, but solis-sensor without control part enabled.
I enabled my integration yesterday, and until now no issues - inverter is fully on-line, no drops in solis-sensor power readings.

jmason, 2025-03-31:
Can anyone seeing this issue please try out the patch in https://github.com/hultenvp/solis-sensor/pull/441 to see if that helps?  I think it's likely to fix it...

sjoram, 2025-03-31:
OK - nooby question - how do I install the patch? Do I copy the files directly into my install and restart or is there an automated way?

jmason, 2025-03-31:
This should do it:

```
git clone git@github.com:jmason/solis-sensor.git
git checkout jmason/reduce-polling-in-control-mode
```

and then copy the files as described at https://github.com/jmason/solis-sensor/tree/jmason/reduce-polling-in-control-mode?tab=readme-ov-file#manual-installation .

mkuthan, 2025-04-01:
I'm not sure if changing the polling frequency will help. With a single API atRead call for each entity, the integration might still hit the official limit of 2 requests per second.

I solved this issue by using the HA coordinator and an undocumented endpoint, atReadBatch, for reading the state of entities at once.