hultenvp/solis-sensor #438: Solis Integration hanging during start up and seemingly crashing HA

sjoram, 2025-03-20:
I had some issues over the week and just recreated my HA container pulling the latest version as of today (March 20th 2025).

My HA instance is now trying to start but hangs during the SOLIS integration start up and then returns 'Connection Lost. Reconnecting' so it is basically not starting.

I can't find anything untoward in the logs as I guess it just bails out before it can write anything!

Have I made a mistake upgrading to the latest HA version? Should I revert or is there something else I am missing and can do to get it to start?

Many thanks

sjoram, 2025-03-20:
Actually, scratch that - after multiple reboots and container restarts the Solis Integration is actually starting now albeit very, very slowly!

However, it does not appear to be connecting and it is not returning any data after 20 minutes.

hultenvp, 2025-03-24:
There is no "connection lost" message in the solis code.
Can you still share the debug log, even though it does not say much?

sjoram, 2025-03-24:
The browser is losing connection to the HA instance. This is while HA is trying to set up the Solis integration. There is no way then forward except to try to restart HA and hope it makes it through the next time as HA stops serving.
After many, many restarts HA made it through. However, I then saw my Solis dongle had stopped communicating so I had no data and got a lot of Attempt Failed log entries as below. This  communication stopping is unfortunately happening every few days now since I updated to the control integration. If I knew how to disable that I would! The log is huge but here are a few snippets.

```
2025-03-20 16:44:07.391 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-03-20 16:44:07.393 DEBUG (MainThread) [custom_components.solis.service] Starting controls discovery
2025-03-20 16:44:07.394 DEBUG (MainThread) [custom_components.solis.soliscloud_api] No firmware version found for Inverter SN 6031050232111211
2025-03-20 16:44:07.665 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Unable to determine HMI firmware version for Inverter SN 603105....................
2025-03-20 16:44:07.665 DEBUG (MainThread) [custom_components.solis.service] Inverter SN 60310.................... HMI status None
2025-03-20 16:44:07.666 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 161, in async_discover
    await self._discover_controls(inverter_serials)
  File "/config/custom_components/solis/service.py", line 182, in _discover_controls
    control_desciptions = ALL_CONTROLS[hmi_flag]
                          ~~~~~~~~~~~~^^^^^^^^^^
KeyError: None
2025-03-20 16:44:10.440 DEBUG (MainThread) [custom_components.solis.select]     Attempt 0 failed
2025-03-20 16:44:10.441 DEBUG (MainThread) [custom_components.solis.number]     Attempt 0 failed
2025-03-20 16:44:10.443 DEBUG (MainThread) [custom_components.solis.time]     Attempt 0 failed
2025-03-20 16:44:10.444 DEBUG (MainThread) [custom_components.solis.button]     Attempt 0 failed
```

```
2025-03-20 16:44:12.440 WARNING (MainThread) [homeassistant.components.select] Setup of select platform solis is taking over 10 seconds.
2025-03-20 16:44:12.441 WARNING (MainThread) [homeassistant.components.number] Setup of number platform solis is taking over 10 seconds.
2025-03-20 16:44:12.443 WARNING (MainThread) [homeassistant.components.time] Setup of time platform solis is taking over 10 seconds.
2025-03-20 16:44:12.444 WARNING (MainThread) [homeassistant.components.button] Setup of button platform solis is taking over 10 seconds.
2025-03-20 16:44:20.443 DEBUG (MainThread) [custom_components.solis.select]     Attempt 1 failed
2025-03-20 16:44:20.444 DEBUG (MainThread) [custom_components.solis.number]     Attempt 1 failed
2025-03-20 16:44:20.444 DEBUG (MainThread) [custom_components.solis.time]     Attempt 1 failed
2025-03-20 16:44:20.445 DEBUG (MainThread) [custom_components.solis.button]     Attempt 1 failed
2025-03-20 16:44:30.445 DEBUG (MainThread) [custom_components.solis.select]     Attempt 2 failed
2025-03-20 16:44:30.445 DEBUG (MainThread) [custom_components.solis.number]     Attempt 2 failed
2025-03-20 16:44:30.446 DEBUG (MainThread) [custom_components.solis.time]     Attempt 2 failed
2025-03-20 16:44:30.446 DEBUG (MainThread) [custom_components.solis.button]     Attempt 2 failed
2025-03-20 16:44:40.447 DEBUG (MainThread) [custom_components.solis.select]     Attempt 3 failed
2025-03-20 16:44:40.447 DEBUG (MainThread) [custom_components.solis.number]     Attempt 3 failed
2025-03-20 16:44:40.447 DEBUG (MainThread) [custom_components.solis.time]     Attempt 3 failed
2025-03-20 16:44:40.448 DEBUG (MainThread) [custom_components.solis.button]     Attempt 3 failed
2025-03-20 16:44:49.210 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: {('solis', '81ef074680babe0.......................'): 5721.395010391}
2025-03-20 16:44:50.448 DEBUG (MainThread) [custom_components.solis.select]     Attempt 4 failed
2025-03-20 16:44:50.449 DEBUG (MainThread) [custom_components.solis.number]     Attempt 4 failed
2025-03-20 16:44:50.449 DEBUG (MainThread) [custom_components.solis.time]     Attempt 4 failed
2025-03-20 16:44:50.449 DEBUG (MainThread) [custom_components.solis.button]     Attempt 4 failed
2025-03-20 16:45:00.450 DEBUG (MainThread) [custom_components.solis.select]     Attempt 5 failed
2025-03-20 16:45:00.450 DEBUG (MainThread) [custom_components.solis.number]     Attempt 5 failed
2025-03-20 16:45:00.451 DEBUG (MainThread) [custom_components.solis.time]     Attempt 5 failed
2025-03-20 16:45:00.451 DEBUG (MainThread) [custom_components.solis.button]     Attempt 5 failed
2025-03-20 16:45:02.440 ERROR (MainThread) [homeassistant.components.select] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2025-03-20 16:45:02.441 ERROR (MainThread) [homeassistant.components.number] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2025-03-20 16:45:02.444 ERROR (MainThread) [homeassistant.components.time] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2025-03-20 16:45:02.445 ERROR (MainThread) [homeassistant.components.button] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2025-03-20 16:45:02.451 INFO (MainThread) [homeassistant.bootstrap] Home Assistant initialized in 78.75s
```

patduckless, 2025-03-27:
I take it you are running HA in Docker. What is the Docker HA container showing at startup?

deligatedgeek, 2025-04-23:
I have the following logs for the same issue
`homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:52:58.777 WARNING (MainThread) [homeassistant.components.select] Setup of select platform solis is taking over 10 seconds.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:52:58.777 WARNING (MainThread) [homeassistant.components.number] Setup of number platform solis is taking over 10 seconds.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:52:58.778 WARNING (MainThread) [homeassistant.components.time] Setup of time platform solis is taking over 10 seconds.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:52:58.778 WARNING (MainThread) [homeassistant.components.button] Setup of button platform solis is taking over 10 seconds.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:53:48.776 ERROR (MainThread) [homeassistant.components.select] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:53:48.777 ERROR (MainThread) [homeassistant.components.number] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:53:48.777 ERROR (MainThread) [homeassistant.components.time] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
homeassistant.1.gxdxmvgfc9xy@home-esx-swarm-02.thelabratorium.cloudns.cl    | 2025-04-22 15:53:48.777 ERROR (MainThread) [homeassistant.components.button] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
`

patduckless, 2025-06-19:
I have done some investigating and it seems like if inverter control is enabled it takes longer for the integration to start. Not too sure why this is at the moment.

AlexZuiderwijk, 2025-07-08:
Same issue here since 2 days.....

```
2025-07-09 00:17:21.129 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['110792215310015']
2025-07-09 00:17:21.129 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110792215310015
2025-07-09 00:17:23.270 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-07-09 00:17:23.270 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Valid inverters: ['110792215310015']
2025-07-09 00:17:23.301 DEBUG (MainThread) [custom_components.solis.soliscloud_api] CSRF token acquired
2025-07-09 00:17:23.302 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110792215310015
2025-07-09 00:17:25.406 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-07-09 00:17:25.407 DEBUG (MainThread) [custom_components.solis.service] Starting controls discovery
2025-07-09 00:17:25.407 DEBUG (MainThread) [custom_components.solis.soliscloud_api] No firmware version found for Inverter SN 110792215310015
2025-07-09 00:17:25.496 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Unable to determine HMI firmware version for Inverter SN 110792215310015
2025-07-09 00:17:25.497 DEBUG (MainThread) [custom_components.solis.service] Inverter SN 110792215310015 HMI status None
2025-07-09 00:17:25.497 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 163, in async_discover
    await self._discover_controls(inverter_serials)
  File "/config/custom_components/solis/service.py", line 191, in _discover_controls
    control_desciptions = ALL_CONTROLS[hmi_flag]
                          ~~~~~~~~~~~~^^^^^^^^^^
2025-07-09 00:17:27.956 DEBUG (MainThread) [custom_components.solis.select]     Attempt 0 failed
2025-07-09 00:17:27.956 DEBUG (MainThread) [custom_components.solis.number]     Attempt 0 failed
2025-07-09 00:17:27.958 DEBUG (MainThread) [custom_components.solis.time]     Attempt 0 failed
2025-07-09 00:17:27.959 DEBUG (MainThread) [custom_components.solis.button]     Attempt 0 failed
2025-07-09 00:17:29.955 WARNING (MainThread) [homeassistant.components.select] Setup of select platform solis is taking over 10 seconds.
2025-07-09 00:17:29.956 WARNING (MainThread) [homeassistant.components.number] Setup of number platform solis is taking over 10 seconds.
2025-07-09 00:17:29.957 WARNING (MainThread) [homeassistant.components.time] Setup of time platform solis is taking over 10 seconds.
2025-07-09 00:17:29.958 WARNING (MainThread) [homeassistant.components.button] Setup of button platform solis is taking over 10 seconds.
2025-07-09 00:17:37.958 DEBUG (MainThread) [custom_components.solis.select]     Attempt 1 failed
2025-07-09 00:17:37.959 DEBUG (MainThread) [custom_components.solis.number]     Attempt 1 failed
2025-07-09 00:17:37.959 DEBUG (MainThread) [custom_components.solis.time]     Attempt 1 failed
2025-07-09 00:17:37.960 DEBUG (MainThread) [custom_components.solis.button]     Attempt 1 failed

```

After 999 attempts it stops retrying.
Not sure ig the messages "No firmware version found for Inverter", "Unable to determine HMI firmware version for Inverter" and "HMI status None" were also there when the integration was running fine.

BTW, during the 999 attempts every now and then there is also a message like this:
DEBUG (MainThread) [custom_components.solis.number] No controls found for Plant ID 1298491919448774280
DEBUG (MainThread) [custom_components.solis.time] No time found for Plant ID 1298491919448774280

imc67, 2025-07-10:
Since today I have exactly the same issue and logs, result:  the integration doesn't work anymore. That's a pity because of negative energy prices I used this wonderfull plugin the automate the switching off

imc67, 2025-07-12:
Was anyone able to solve this issue? Still not working. 😣

hultenvp, 2025-08-22:
> 2025-07-09 00:17:25.407 DEBUG (MainThread) [custom_components.solis.soliscloud_api] No firmware version found for Inverter SN 110792215310015

This might be the trigger?