hultenvp/solis-sensor #439: control stops working after pyscript update to 1.6.2

eddysteurs, 2025-03-22:
**Describe the bug**
control stops working after applying pyscript update to 1.6.2 
PS : same problem for solis.api

Exception in <file.solis_control.solis_control> line 117: inverterList = response.json() ^ AttributeError: '_BaseRequestContextManager' object has no attribute 'json'

**Config**
HA 2025.3.4 

**Versions**
HA 2025.3.4 

**Debug logs**
Add to configuration.yaml:
Logger: custom_components.pyscript.file.solis_control.solis_control
Source: custom_components/pyscript/eval.py:511
integration: Pyscript Python scripting (documentation, issues)
First occurred: 12:16:58 (5 occurrences)
Last logged: 12:24:13
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

hultenvp, 2025-03-24:
@jmason: Assigning to you, leave it to you if you want people to issue bugs in your fork or here.

jmason, 2025-03-24:
@hultenvp my fork is only used to stage PRs for any potential changes that need to be merged back, so probably best to keep any bugs here!

jmason, 2025-03-24:
It looks like pyscript 1.6.3 has just been released to fix this issue; see https://github.com/custom-components/pyscript/issues/710 -- could you try that?

jmason, 2025-03-26:
going to close this report on the assumption that was it.  Feel free to reopen if not.

closed 2025-03-26 by jmason