hultenvp/solis-sensor #440: Enhancement - Turn off/on inverter in Home assistant

J3G0, 2025-03-26:
**Describe the bug**
Feature request. The Solis cloud portal allows you to turn on or off the inverter.

I would like to request this to be available in Home Assistant. This would enable automations. This way, when prices are negative, or batteries are full, we can turn off the solar production to prevent paying money to deliver power to the grid.

jmason, 2025-03-26:
I guess that is this setting in the Soliscloud control page:

[image]

right?

J3G0, 2025-03-26:
I assume? I checked it in the app, there it is listed under the inverter settings.

disxount, 2025-04-05:
Will this be added in next release?

skrachs, 2025-04-05:
Wouldn't it be better to limit export power rather than turn off the inverter?
Is it possible to adjust the export power limit with this integration?

J3G0, 2025-04-05:
> Wouldn't it be better to limit export power rather than turn off the inverter? Is it possible to adjust the export power limit with this integration?

The integration is limited to the possibilities available on solis cloud as that's the connection. It's also via solis cloud servers and not "offline".

Limiting the export is not what you want. If the price is positive you want to produce as much as you can. Of the price is negative you do not want to export anything.

Throttling export could be interesting when you still want to use your own solar. But with negative proces you want to consume as much power from the grid as possible, as then you get paid to consume power from the grid.

skrachs, 2025-04-06:
It is what I want to do when the price is negative.
I don't want to export and I can't consume all energy.

So is it possible to add this setting to the integration?

[image]

disxount, 2025-04-06:
This thread is for the function on / off, please add extra request maybe

For dynamic energy prices on / off is the best option.

J3G0, 2025-04-06:
> It is what I want to do when the price is negative. I don't want to export and I can't consume all energy.
> 
> So is it possible to add this setting to the integration?
> 
> [image]

FYI This is for battery solis systems. Not sure if the inverter itself can be throttled. Would be helpful when you have a different brand energy storage.

On/off would already be very helpful. So happy to keep this tread just for the on/off request indeed.

J3G0, 2025-04-27:
Today I will manually turn of the inverter in the soliscloud app. We will have 3 hours of negative prices. With this switch visible in home assistant we can easily implement an automation to switch off during this time. Looking forward to having this implemented :)

[image]

plebcity, 2025-05-04:
The PR is merged, but we probably need to fix #452 first before anybody can use it.

damiancosta82, 2025-05-05:
Have you been able to control the on/off of the inverter? I am trying to find a switch to automatize the inverter with the dynamic tariff.

J3G0, 2025-05-05:
In Solis cloud or the Solis cloud app, yes. But it is not implemented yet in the HACS repo. See above.

damiancosta82, 2025-05-05:
And when can be implemented? I would like to automatise this switch with the dynamic price.

plebcity, 2025-05-05:
> And when can be implemented? I would like to automatise this switch with the dynamic price.El

I have already implemented it, we just need to wait on the maintainer of this repository to release a new version of the integration

disxount, 2025-05-08:
It is the : Solis Inverter energy export on/off Control in HA

[image]

[image]

damiancosta82, 2025-05-08:
Hi,

I don't get these commands:

[cid:c41193fa-f848-4658-b0e6-ea9e7c34f72b]
[cid:6640d787-9fe6-47f4-80df-229773ecc993]

[cid:4341b925-3e76-4630-8a48-d8f2e750c182]

image.png (view on web)<https://github.com/user-attachments/assets/75adafcc-1f6c-41e5-b9b6-ad1ebe677e64>

image.png (view on web)<https://github.com/user-attachments/assets/be67a9db-3bcb-4112-a944-812f38cf1df3>

closed 2025-05-08 by J3G0

damiancosta82, 2025-05-08:
> It didn't show up for me directly. It worked for me once I reconfigured the integration entries. So I will close this ticket.

How do you reconfigure the integration entries? I have remove the integration and re-installed, but it is still the same.

J3G0, 2025-05-08:
Check that you have the correct version and click configure. You might need a reboot. 

[image]

damiancosta82, 2025-05-08:
Hi,

I have the latest version and I have reconfigured 3 times. I only get 88 entities when control is activated. Which inverter do you have?

[cid:d22f7789-891e-472f-ba7f-e6e51073c1b1]

Screenshot_20250508-153447.png (view on web)<https://github.com/user-attachments/assets/a9b22adc-c40e-41a3-9b51-9d39a5a47d51>

plebcity, 2025-05-08:
> Hi,
> 
> I have the latest version and I have reconfigured 3 times. I only get 88 entities when control is activated. Which inverter do you have?
> 
> [cid:d22f7789-891e-472f-ba7f-e6e51073c1b1]
> […](#)

You need a S3-WIFI-ST datalogger or better, check your datalogger model

plebcity, 2025-05-08:
> I have S3-WIFI-ST datalogger, I can manage the inverter via soliscloud, but the switch to on/off the inverter is not coming through, could it be that they have a different ID for different models?
> 
> Below is a picture of my soliscloud
> 
> [cid:403518f5-7909-4852-9348-39fb508bdf6a]
> […](#)

Can you turn the inverter off in the soliscloud interface? If you can can you post the response your get from the call to the soliscloud server like I did in the PR linked to this issue?

damiancosta82, 2025-05-08:
Here you have
This is for switching off
[cid:a79dbed3-8d20-4751-bf9c-10b08646e9b9]

This is for switching on

[cid:3e278fa5-7e01-430f-906f-654f373e3af0]

damiancosta82, 2025-05-08:
Yes, please see them here
This is for switching off

[image]

This is for switching on
[image]

plebcity, 2025-05-08:
> Yes, please see them here This is for switching off
> 
> [image]
> 
> This is for switching on [image]

I think you are looking at the wrong settings. Go to your inverter, then on the top right corner go to inverter control and then you see a screen like this and then you can turn the export on and off.

[image]

damiancosta82, 2025-05-08:
Do you mean this one?

[image]

damiancosta82, 2025-05-08:
here in english:

[image]

plebcity, 2025-05-08:
> Do you mean this one?

Please go to inspect element and select the network tab, then in the soliscloud interface click on the on/off menu on the left again. You will see a request called atReadBatch in your network tab, click on it and validate if the response you get has the id 5161 under sysCommand.

[image]

damiancosta82, 2025-05-08:
Sorry but could you please guide me a little, inspect element is in solidcloud or in HA?

plebcity, 2025-05-08:
> Sorry but could you please guide me a little, inspect element is in solidcloud or in HA?

Soliscloud, I want to know if the ids are the issue or there is something else that is the issue

damiancosta82, 2025-05-08:
Thanks, I get this

[image]

plebcity, 2025-05-08:
> Thanks, I get this

How strange, you have 5162 as the id for the on/off switch while I have 5161. Maybe it has something to do with inverterModelID or something. Maybe on of the other contributors knows if we had any issues like this before with the other controls? (@jmason  @patduckless or @hultenvp )

damiancosta82, 2025-05-08:
I have change the ID in the file editor, but it is not picking up the switch in HA, do I need to make any other change?

[image]

plebcity, 2025-05-08:
> I have change the ID in the file editor, but it is not picking up the switch in HA, do I need to make any other change?

Can you enable debug logging in HA for the soliscloud integration and then restart home assistant? Please post the debug logging of the soliscloud integration here. It might have something to do with the HMI version or something like that.

damiancosta82, 2025-05-08:
These are the key difference between your log and mine:

[image]

damiancosta82, 2025-05-09:
Please see the log attached

[Log HA.txt](https://github.com/user-attachments/files/20116874/Log.HA.txt)

plebcity, 2025-05-09:
> Please see the log attached

Thanks, seems like the HMI version is lower than what I have. I can make a new MR to add the on/off switch for your HMI version using the id 5162

plebcity, 2025-05-09:
> These are the key difference between your log and mine:

In this JSON response there is a list of the options you can select in different languages, I need the corresponding id of the ON and OFF option. On my inverter ON is 190 and OFF is 222. Could you please take a look at that and provide the correct id's for the ON and OFF option?

damiancosta82, 2025-05-09:
That will be great

damiancosta82, 2025-05-09:
I have checked and it is the same, On is 190 and OFF is 222.

damiancosta82, 2025-05-10:
Do you have any update?

plebcity, 2025-05-10:
> Do you have any update?

The PR is merged, now waiting for @hultenvp to make a new release

plebcity, 2025-05-10:
It's working just in time, tomorrow is gonna be even worse than today. Today it was €-0,01 between 13:00 and 14:00, tomorrow is gonna be €-0,25 between 12:00 and 15:00.

[image]

patduckless, 2025-05-10:
Do you have an automation pulling in the price data and switching accordingly?

plebcity, 2025-05-11:
> Do you have an automation pulling in the price data and switching accordingly?

I get the price data from my energy supplier, they have an integration in HA core. I have 2 automations on the price, 1 for below 0 to turn it off and above 0 to turn it back on

damiancosta82, 2025-05-11:
The update is not yet ready in HACS, is there anyway to get it update before tomorrow?

imc67, 2025-05-11:
> > Do you have an automation pulling in the price data and switching accordingly?
> 
> I get the price data from my energy supplier, they have an integration in HA core. I have 2 automations on the price, 1 for below 0 to turn it off and above 0 to turn it back on

Very curious how how you did the automation as mine (create with ChatGPT) today didn’t work in the last step: switch OFF, this is my automation YAML:

````
alias: Zonnepanelen uit bij lage stroomprijs (≤ 0 euro)
description: ""
triggers:
  - entity_id: sensor.energyzero_today_energy_current_hour_price
    below: 0.01
    for:
      minutes: 10
    trigger: numeric_state
conditions:
  - condition: state
    entity_id: select.solis_inverter_energy_export_on_off_control_switch
    state: "ON"
actions:
  - target:
      entity_id: select.solis_inverter_energy_export_on_off_control_switch
    data:
      option: "OFF"
    action: select.select_option
  - data:
      message: "Zonnepanelen uitgeschakeld: stroomprijs is 0 euro of lager."
    action: notify.notify
mode: single
````

plebcity, 2025-05-11:
> Very curious how how you did the automation as mine (create with ChatGPT) today didn’t work in the last step: switch OFF, this is my automation YAML:

Like this, keep in mind that the automation only triggers when the price moves either up or down from the previous value to below or above your trigger. So mine doesn't trigger all the time, just once the price moves below 0,00001

[image]

imc67, 2025-05-11:
> Like this, keep in mind that the automation only triggers when the price moves either up or down from the previous value to below or above your trigger. So mine doesn't trigger all the time, just [...]

Thanks for the inspiration, I changed the Action: select to Select, let’s see in the afternoon if it switches on again.

damiancosta82, 2025-05-12:
It works perfectly, thanks a lot.

plebcity, 2025-05-12:
> It works perfectly, thanks a lot.

Good to hear, no worries. Does the power limit setting I added also work for you? I guessed the id would be the same but not sure.

cafedelmar01, 2025-05-12:
Hi All, 

Does this only works with as particulair set off inverters?

Mine reads all the values, and is able to set all the time values, 

But is doesn't seems to forward the commands when the export-values are adjusted:

Model: S5-EH1P5K-L 
FW: 4C0050-650000

damiancosta82, 2025-05-13:
How is called the entity? I could not find it.

plebcity, 2025-05-13:
> How is called the entity? I could not find it.

Solis Power Limit Setting, in Soliscloud its in the Inverter Power Setting menu. Its a percentage of the rated power, setting it to 10% will limit the inverter output to 10% of its rated power.

damiancosta82, 2025-05-13:
I tried but it does not work, I could not find that setting in soliscloud, where is it?

damiancosta82, 2025-05-13:
Hi Koen,

Now I have the two commands in HA, but they do not change the status of the inverter in soliscloud? do you know what could it be?

damiancosta82, 2025-05-13:
I also realised that the commands are swtiched for 5162 OFF is 190 and ON is 222. I have changed that in the file control_const.py but it is not changing anything in soliscloud.

KR

plebcity, 2025-05-13:
> Hi Koen,
> 
> Now I have the two commands in HA, but they do not change the status of the inverter in soliscloud? do you know what could it be?

I added the Solis Power Limit Setting yesterday, if that is not working for you then it might not be available in the Soliscloud API for your inverter or the id might be different just like the ON/OFF switch was different.

plebcity, 2025-05-13:
> I also realised that the commands are swtiched for 5162 OFF is 190 and ON is 222. I have changed that in the file control_const.py but it is not changing anything in soliscloud.

So its different compared to what you said here? https://github.com/hultenvp/solis-sensor/issues/440#issuecomment-2865461063

patduckless, 2025-05-13:
@damiancosta82 after you made the change to the control_cost.py did you restart home assistant?

plebcity, 2025-05-13:
[image]

Power limit setting is working perfectly for me, I used the following script to automate the limit percentage:

```
description: Set a limit for the inverter based on current energy usage
variables:
  max_percentage: 110
  max_power: 4400
sequence:
  - variables:
      current_percentage: "{{ states('number.solis_power_limit_setting') | float(0) }}"
      current_power: >-
        {{ states('sensor.p1_meter_5c2faf0a48fe_actueel_vermogen') | float(0) +
        50 }}
      raw_new_percentage: |
        {{ (current_power / max_power * max_percentage) + current_percentage }}
      new_percentage: |
        {{ [raw_new_percentage, max_percentage] | min | round(0) }}
  - if:
      - condition: template
        value_template: |
          {{ current_power < -40 or current_power > 20 }}
    then:
      - target:
          entity_id: number.solis_power_limit_setting
        data:
          value: "{{ new_percentage }}"
        action: number.set_value

```

It will set a new percentage if the current power is above -30W or below -90W, this way I make sure that it (almost) never goes above 0W.
This script runs every 5 seconds if the export price is negative

[8 more replies]